zephyrproject-rtos/zephyr #28460: Generated ExternalProject include directories

pablogs9, 2020-09-17:
Hello, I'm developing a Zephyr module that generates a static library and some headers using `externalproject_add`. 
The point is that I want to link applications against this library but it raises an error about not finding headers, because they are not generated yet. The application is built before the ExternalProject so the second time I run `west build` everything works.

My module CMakeLists.txt looks like this:

```cmake
...
include(ExternalProject)

set(INSTALLATION_DIR [ABSOLUTE PATH TO MY LIBRARY INSTALLATION])

externalproject_add(mylib_project
    PREFIX     ${CMAKE_BINARY_DIR}/mylib-prefix
    SOURCE_DIR ${INSTALLATION_DIR}
    BINARY_DIR ${INSTALLATION_DIR}
    CONFIGURE_COMMAND ""
    BUILD_COMMAND
             # MAKE BUILD COMMAND HERE
    INSTALL_COMMAND ""
    BUILD_BYPRODUCTS ${INSTALLATION_DIR}/mylib.a
    )

zephyr_library()
zephyr_library_sources("additional_source_code.c")
zephyr_include_directories(${INSTALLATION_DIR}/include)
zephyr_link_libraries(gcc)
zephyr_library_import(mylib ${INSTALLATION_DIR}/mylib.a)
...
```

This approach generates the `include` and `mylib.a` after my app code has failed with:

```
main.c:12:10: fatal error: mylib/my_lib_header.h: No such file or directory
   12 | #include <mylib/my_lib_header.h>
      |          ^~~~~~~~~~~
```

because this `include/mylib` and the `mylib.a` are generated afterwards.

Is there any option for telling Zephyr to resolve this ExternalProject before trying to build my application? My working tree is like this:

```
mylib_app/
├── CMakeLists.txt
├── modules
│   └── mylib
│       ├── CMakeLists.txt
│       ├── [MORE LIBRARY FILES]
│       ├── Kconfig
│       ├── additional_source_code.c
│       ├── zephyr
│       │   └── module.yml
│       └── mylib.mk
├── prj.conf
└── src
    └── main.c
```

And my application CMakeLists.txt is:

```cmake
cmake_minimum_required(VERSION 3.13.1)

list(APPEND ZEPHYR_EXTRA_MODULES ${CMAKE_CURRENT_SOURCE_DIR}/modules/mylib)
find_package(Zephyr REQUIRED HINTS $ENV{ZEPHYR_BASE})
project(mylib_app)
target_sources(app PRIVATE src/main.c)
```

tejlmand, 2020-09-18:
@pablogs9 Try to link the library instead, that should ensure correct dependency handling in CMake:
```
zephyr_library_link_libraries(${INSTALLATION_DIR}/mylib.a)
```

tejlmand, 2020-09-18:
alternatively, use `add_dependencies()` directly, as
```
add_dependencies(${ZEPHYR_CURRENT_LIBRARY} mylib_project)
```

pablogs9, 2020-09-18:
Hello @tejlmand, thanks for your answer, I have solved it with:

```cmake
...
include(ExternalProject)

externalproject_add(libmicroros_project
[...]
    )

zephyr_library_import(libmicroros ${MICROROS_DIR}/libmicroros.a)

zephyr_interface_library_named(microros)
add_dependencies(microros libmicroros)
add_dependencies(microros libmicroros_project)
target_include_directories(microros INTERFACE ${MICROROS_DIR}/include)
...
```

and including a Kconfig variable:

```kconfig
menuconfig MICROROS
    bool
	default n
    prompt "Enables a micro-ROS library"
    select APP_LINK_WITH_MICROROS
    help
        Enables a micro-ROS library

config APP_LINK_WITH_MICROROS
    bool "Link 'app' with MICROROS"
    default n
    help
        Add MICROROS header files to the 'app' include path. 
[...]
```
I guess that making `microros` depending on `libmicroros_project` and the app depending on `microros` (because of this `APP_LINK_WITH_MICROROS`)  makes the build system build and configure my external library just before the application.

For me this solution it's ok, but I would like to have a detailed explanation about this `APP_LINK_WITH_` flag because I have not found documentation about when modules are (or are not) linked against the user app.

Thank you very much!

tejlmand, 2020-09-18:
> Hello @tejlmand, thanks for your answer, I have solved it with:

Great, and for a reusable module, that can be configured through Kconfig, that is the cleanest solution.
I just presented the short hand version, for an external module where nothing else is needed.

> For me this solution it's ok, but I would like to have a detailed explanation about this `APP_LINK_WITH_` flag because I have not found documentation about when modules are (or are not) linked against the user app.

The best description currently available, can be found directly in the extensions file
https://github.com/zephyrproject-rtos/zephyr/blob/master/cmake/extensions.cmake#L528-L550

I still have a task on a too full plate to get the Zephyr CMake extensions and Kconfig integration better described in the regular documentation.

closed 2020-09-18 by tejlmand