zephyrproject-rtos/zephyr #28461: `HCI_CMD_TIMEOUT` when setting ext adv data in the hci_usb project

Thalley, 2020-09-17:
**Describe the bug**
Attempted to use the `samples/bluetooth/hci_usb/` project with `CONFIG_BT_CTLR_ADV_EXT=y`, to run the BT shell as a native_posix application over HCI over USB for on a NRF52840-DK. This unfortunately results in a `HCI_CMD_TIMEOUT` when setting the data, as part of the `bt advertise on` command in the shell. 

It works as intended if flashing the shell project directly to the board, but not when run over USB. 

**To Reproduce**
Steps to reproduce the behavior:
1. Build `samples/bluetooth/hci_usb/` with `CONFIG_BT_CTLR_ADV_EXT=y` and flash to board
2. Build `tests/bluetooth/shell/` with the `native_posix` board
3. Run the shell and enter `bt init` followed by `bt advertise on`. 
4. See error

**Expected behavior**
Expected to start advertising legacy adv PDUs but with the extended adv HCI commands. 

**Impact**
Inconvenience. 

**Logs and console output**
Shell log
```
bt init
[00:00:04.610,000] <wrn> bt_hci_core: ECC HCI commands not available
[00:00:04.610,000] <inf> bt_hci_core: Identity: d6:2c:51:b0:60:7a (random)
[00:00:04.610,000] <inf> bt_hci_core: HCI: version 5.2 (0x0b) revision 0x0000, manufacturer 0x05f1
[00:00:04.610,000] <inf> bt_hci_core: LMP: version 5.2 (0x0b) subver 0xffff
Bluetooth initialized
uart:~$ bt advertise on
[00:00:17.620,000] <err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
[00:00:17.620,000] <err> os: Current thread: 0x5676a800 (shell_uart)
[00:00:17.620,000] <err> os: Halting system
```
BTMON log
```
< HCI Command: LE Set Extended Advertising Parameters (0x08|0x0036) plen 25      #2247 12009.382438
        Handle: 0x00
        Properties: 0x0013
          Connectable
          Scannable
          Use legacy advertising PDUs: ADV_IND
        Min advertising interval: 100.000 msec (0x00a0)
        Max advertising interval: 150.000 msec (0x00f0)
        Channel map: 37, 38, 39 (0x07)
        Own address type: Random (0x01)
        Peer address type: Public (0x00)
        Peer address: 00:00:00:00:00:00 (OUI 00-00-00)
        Filter policy: Allow Scan Request from Any, Allow Connect Request from Any (0x00)
        TX power: 127 dbm (0x7f)
        Primary PHY: LE 1M (0x01)
        Secondary max skip: 0x00
        Secondary PHY: Reserved (0x00)
        SID: 0x00
        Scan request notifications: Disabled (0x00)
> HCI Event: Command Complete (0x0e) plen 5                                      #2248 12009.383167
      LE Set Extended Advertising Parameters (0x08|0x0036) ncmd 1
        Status: Success (0x00)
        TX power (selected): 127 dbm (0x7f)
< HCI Command: LE Set Advertising Set Random Address (0x08|0x0035) plen 7        #2249 12009.392330
        Advertising handle: 0x00
        Advertising random address: FA:F5:EA:1C:62:4C (OUI FA-F5-EA)
> HCI Event: Command Complete (0x0e) plen 4                                      #2250 12009.393198
      LE Set Advertising Set Random Address (0x08|0x0035) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Extended Advertising Data (0x08|0x0037) plen 255           #2251 12009.422305
        Handle: 0x00
        Operation: Complete extended advertising data (0x03)
        Fragment preference: Minimize fragmentation (0x01)
        Data length: 0x07
        Flags: 0x06
          LE General Discoverable Mode
[... 642 lines cut ...]
        Fragment preference: Minimize fragmentation (0x01)
        Data length: 0x03
        Flags: 0x04
          BR/EDR Not Supported
@ MGMT Event: Command Complete (0x0001) plen 7                                {0x0001} 12019.489290
      Set Powered (0x0005) plen 4
        Status: Success (0x00)
        Current settings: 0x00008a11
          Powered
          Bondable
          Low Energy
          Secure Connections
          Static Address
> HCI Event: Command Complete (0x0e) plen 4                                      #2299 12019.490038
      LE Set Extended Advertising Data (0x08|0x0037) ncmd 1
        Status: Success (0x00)
```

**Environment (please complete the following information):**
 - OS: linux 5.8.9-arch2-1
 - Toolchain: Zephyr SDK 0.11.3
 - Commit SHA: 0076313bcb5a75151c6741b0e252f182568a51ee

**Additional context**
As mentioned above, the issue only seems to exists when running with HCI over USB.

joerchan, 2020-09-18:
@Thalley I'm guessing buffer size configurations are not right, especially since you crash on the ext adv data.
Can you enable CONFIG_BT_EXT_ADV in hci_uart as well?
Ideally you would only have to enable that one, but for the moment we have this:
```
	# Enable by default for BT_LL_SW_SPLIT when "LE Advertising Set Terminated event" is implemented
	default y if BT_EXT_ADV && !BT_LL_SW_SPLIT
```

emob-nordic, 2020-09-21:
@Thalley Have You tried with `CONFIG_BT_EXT_ADV` as @joerchan suggested, does it work for You?

Thalley, 2020-09-21:
@emob-nordic @joerchan Not yet, been busy with some issue with Periodic Advertisements. Will try to get around to it tonight.

Thalley, 2020-09-21:
@emob-nordic @joerchan Adding `CONFIG_BT_EXT_ADV` to `hci_usb` (don't have the cables at home to run over UART) gives the same result. 

If there's any buffer size configurations I should try, please let me know which :)

emob-nordic, 2020-09-23:
Please try to set `CONFIG_USB_REQUEST_BUFFER_SIZE` to 384 for instance. That fixed the issue for me.

Thalley, 2020-09-23:
> Please try to set `CONFIG_USB_REQUEST_BUFFER_SIZE` to 384 for instance. That fixed the issue for me.

@emob-nordic That works for me too; wouldn't have guess that the USB buffer would be an issue on a native_posix build :D

closed 2020-09-29 by ioannisg

Thalley, 2020-10-01:
@emob-nordic @joerchan 
Using the same setup (i.e. the `samples/bluetooth/hci_usb/` with `CONFIG_BT_CTLR_ADV_EXT=y`) I still experience the same issue. 

The PR (https://github.com/zephyrproject-rtos/zephyr/pull/28637) sets the HCI buffer size if `BT_RX_BUF_LEN > 76` which it will only be by default if e.g. `BT_EXT_ADV` is enabled. 

Setting `BT_EXT_ADV=y` in the prj.conf file would fix this, but I was under the impression that `BT_EXT_ADV` as well as `BT_RX_BUF_LEN` are BT host configurations, and it is somewhat odd that those should have an impact on the the HCI USB project as I see that as a "controller" that exposes a HCI interface over USB. 

Am I under the wrong impressions regarding the above?

joerchan, 2020-10-01:
@Thalley No, those are common configurations, not host specific. BT_RX_BUF_LEN is used by HCI_RAW as well, and BT_RX_BUF_LEN default is determined based on whether BT_EXT_ADV is enabled or not.

You could only set BT_RX_BUF_LEN, but then there are migt be other  buffers you would have to update, so best would be select the BT_EXT_ADV feature.

emob-nordic, 2020-10-01:
@joerchan Shouldn't the `BT_EX_BUF_LEN` select `BT_EXT_ADV` then? Or extended buffer does not always mean extended advertising?

Thalley, 2020-10-01:
@joerchan Should the `BT_RX_BUF_LEN` really be located in subsys/bluetooth/*host*/Kconfig then? 

I see that `BT_CTLR_ADV_EXT` has `default y if BT_EXT_ADV && !BT_LL_SW_SPLIT` so I suppose setting `BT_EXT_ADV` would be sufficient?

joerchan, 2020-10-01:
> Shouldn't the BT_EX_BUF_LEN select BT_EXT_ADV then? Or extended buffer does not always mean extended advertising?

No, the length of the required buffers could be dependent on other features as well, like Data Length Update, maximum MTU, BR/EDR etc.

> Should the BT_RX_BUF_LEN really be located in subsys/bluetooth/host/Kconfig then

When you build HCI_USB you are including hci_raw, so I don't think moving this into the Host makes sense, as you should set this option for the controller-only samples.

> I see that BT_CTLR_ADV_EXT has default y if BT_EXT_ADV && !BT_LL_SW_SPLIT so I suppose setting BT_EXT_ADV would be sufficient?

Not if you have BT_LL_SW_SPLIT=y, then you need to set it explicitly, this is because the BT_CTLR_ADV_EXT feature is still in development, so I wouldn't want to set it default on unless you are prepared to handle a work in progress feature.

Thalley, 2020-10-01:
> Should the BT_RX_BUF_LEN really be located in subsys/bluetooth/host/Kconfig then

@joerchan The `BT_RX_BUF_LEN` is currently located in the host Kconfig, and my comment was regarding whether it would make more sense to put into the common on controller part. 

Either way, I think it I understand the configs better now, and the PR does fix this issue :)

joerchan, 2020-10-01:
@Thalley Sorry, I misread. Yes, perhaps it should be moved, but TBH I'm quite happy to leave it alone right now.

PR #28637 (2020-09-29): usb: bluetooth: Increase data stage buffer for setup transfers.
Depending on the Bluetooth features enabled, the HCI
packets can be 264 Bytes long. For HCI over USB the HCI packet
is handled in data stage of USB transfer. Buffer for the data
stage was too short and USB driver was STALLing the data stage
making it unable to complete. Extend the data stage buffer
to 266 Bytes if HCI USB is selected and BT_RX_BUF_LEN > 127
indicating that longer HCI packets are needed.

Fixes: #28461