zephyrproject-rtos/zephyr #28462: SHIELD not handled correct in CMake when using custom board

lund-prevas, 2020-09-17:
In CMakeList.txt set BOARD_ROOT and SHIELD e.g
set(BOARD_ROOT ${MY_CUSTOM_BOARD_ROOT})
set(SHIELD myShield)
.
pull in the Zephyr boilerplate

CMake will then look for the myShield DTS overlay in ${MY_CUSTOM_BOARD_ROOT}/boards/shields/myShield
but it will look for the Kconfig.shield in .../zephyr/boards/shields/myShield
Furthermore is the SHIELD_AS_LIST only set to myShield, when CMake scans ${MY_CUSTOM_BOARD_ROOT}, it is empty when it scans zephyr/boards

Expected behaviour would be both DTS overlay and Kconfig.shield was found in same location.
First look for ${MY_CUSTOM_BOARD_ROOT}/boards/shields/myShield
and if not found look for .../zephyr/boards/shields/myShield

erwango, 2020-09-17:
@tejlmand would you have time to check this ?

lund-prevas, 2020-10-22:
Any update on this issue ?

tejlmand, 2020-10-27:
@lund-prevas Fixed here: #29571

PR #29571 (2020-11-13): cmake: ensure shields can be placed in other BOARD_ROOTs
Fixes: #28462

This commit allows shields to be defined in other BOARD_ROOTs, either
using `-DBOARD_ROOT=<path>` or a Zephyr moduled defined BOARD_ROOT.
commit 4558ba71 (2020-11-13): cmake: ensure shields can be placed in other BOARD_ROOTs

Fixes: #28462

This commit allows shields to be defined in other BOARD_ROOTs, either
using `-DBOARD_ROOT=<path>` or a Zephyr module defined BOARD_ROOT.
closed 2020-11-13 by carlescufi