zephyrproject-rtos/zephyr #28465: Building OpenThread NCP build system has concurrency issue

markus-becker-tridonic-com, 2020-09-17:
**Describe the bug**

Rarely I do get build errors when building samples/net/openthread/ncp and other openthread samples that also require errno_private.h:

```
In file included from /home/markus/src/zephyrproject/zephyr/lib/libc/minimal/include/errno.h:20,
                 from /home/markus/src/zephyrproject/modules/lib/openthread/src/lib/spinel/spinel.c:48:
/home/markus/src/zephyrproject/zephyr/include/sys/errno_private.h:34:10: fatal error: syscalls/errno_private.h: No such file or directory
   34 | #include <syscalls/errno_private.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

Seems like a concurrency issue in the build system, where the generated file `./build/zephyr/include/generated/syscalls/errno_private.h` is not created by the time spinel.c is being built.

**To Reproduce**
See above.

**Expected behavior**
No concurrency issue.

**Impact**
annoyance. 

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA 04d3c3c1a0d479aada6bab83b3d29e6bf61a2a6a

rlubos, 2020-09-18:
I've run the `ncp` build like several dozen of times, but could not reproduce the error once. @tejlmand Have you seen anything like this before?

tejlmand, 2020-09-18:
I have not seen it, but such randomness could indicate a missing dependency.
If there is a missing dependency, then it can often be triggered with different build commands/arguments, such as:
`ninja / ninja -j1 / ninja j <n> / make / make -j1/ make -j <n> / ...`

will investigate to see if something obviously is missing, otherwise I will try to reproduce the issue.

rlubos, 2020-09-18:
Ok, I've reproduced it, but it appears only to happen with changes from https://github.com/zephyrproject-rtos/zephyr/pull/28458, when RCP variant is built. To reproduce:
`west build -b nrf52840dk_nrf52840 -p -- -DCONF_FILE=prj-rcp.conf`

markus-becker-tridonic-com, 2020-09-21:
@rlubos I have also seen it before RCP.

I was able to reproduce it consistently with:
```
rm -rf build/
export CMAKE_JOB_POOLS=1
west build ....
```

tejlmand, 2020-09-21:
@markus-becker-tridonic-com I haven't been able to reproduce this on https://github.com/zephyrproject-rtos/zephyr/commit/04d3c3c1a0d479aada6bab83b3d29e6bf61a2a6a or master.
Could you check your openthread version ?
```
$ west list openthread
openthread   modules/lib/openthread       7854c687bf23ce509a65d118cbea0c7011a89a1e https://github.com/zephyrproject-rtos/openthread
```

tejlmand, 2020-09-21:
> Ok, I've reproduced it, but it appears only to happen with changes from #28458, when RCP variant is built. To reproduce:
> `west build -b nrf52840dk_nrf52840 -p -- -DCONF_FILE=prj-rcp.conf`

yep, happens on #28458:

It is enough to run:
```
$ cmake -GNinja -DBOARD=nrf52840dk_nrf52840 -Bbuild  samples/net/openthread/ncp/
$ ninja -Cbuild openthread-spinel-rcp -v
...
FAILED: modules/openthread/build/src/lib/spinel/CMakeFiles/openthread-spinel-rcp.dir/spinel.c.obj 
/opt/gcc-arm-none-eabi-7-2018-q2-update/bin/arm-none-eabi-gcc -DMBEDTLS_CONFIG_FILE=\"config-tls-generic.h\" -DOPENTHREAD_CONFIG_BORDER_AGENT_ENABLE=1 -DOPENTHREAD_CONFIG_BORDER_ROUTER_ENABLE=1 -DOPENTHREAD_CONFIG_COMMISSIONER_ENABLE=1 -DOPENTHREAD_CONFIG_DHCP6_SERVER_ENABLE=1 -DOPENTHREAD_CONFIG_ENABLE_BUILTIN_MBEDTLS=0 -DOPENTHREAD_CONFIG_ENABLE_BUILTIN_MBEDTLS_MANAGEMENT=0 -DOPENTHREAD_CONFIG_LOG_LEVEL=0 -DOPENTHREAD_CONFIG_NCP_UART_ENABLE=1 -DOPENTHREAD_CONFIG_THREAD_VERSION=OT_THREAD_VERSION_1_1 -DOPENTHREAD_CONFIG_TMF_NETDATA_SERVICE_ENABLE=1 -DOPENTHREAD_CONFIG_UDP_FORWARD_ENABLE=1 -DOPENTHREAD_ENABLE_NCP=1 -DOPENTHREAD_PROJECT_CORE_CONFIG_FILE=\"openthread-core-zephyr-config.h\" -DOPENTHREAD_RADIO=1 -DOPENTHREAD_SPINEL_CONFIG_OPENTHREAD_MESSAGE_ENABLE=0 -DPACKAGE_NAME=\"OPENTHREAD\" -DPACKAGE_VERSION=\"g2ac83762b\" -Imodules/openthread/build/etc/cmake -I/projects/github/ncs/modules/lib/openthread/etc/cmake -I/projects/github/ncs/modules/lib/openthread/include -I/subsys/net/lib/openthread/platform -I/projects/github/ncs/modules/crypto/mbedtls/configs -I/projects/github/ncs/modules/crypto/mbedtls/include -I/projects/github/ncs/modules/lib/openthread/examples/platforms/zephyr -I/projects/github/ncs/modules/lib/openthread/src -I/projects/github/ncs/modules/lib/openthread/src/core -Os -imacros/projects/github/ncs/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mcpu=cortex-m4 -mthumb -mabi=aapcs -march=armv7e-m -imacros/projects/github/ncs/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -ffunction-sections -fdata-sections -fno-builtin -I/projects/github/ncs/zephyr/include -I/projects/github/ncs/zephyr/build/zephyr/include/generated -I/projects/github/ncs/zephyr/soc/arm/nordic_nrf/nrf52 -I/projects/github/ncs/zephyr/lib/libc/minimal/include -I/projects/github/ncs/zephyr/subsys/settings/include -I/projects/github/ncs/modules/hal/cmsis/CMSIS/Core/Include -I/projects/github/ncs/modules/hal/nordic/drivers/nrf_radio_802154/. -I/projects/github/ncs/modules/hal/nordic/nrfx -I/projects/github/ncs/modules/hal/nordic/nrfx/drivers/include -I/projects/github/ncs/modules/hal/nordic/nrfx/mdk -I/projects/github/ncs/modules/hal/nordic/. -isystem/projects/github/ncs/zephyr/lib/libc/minimal/include   -Wall -Wextra -Wshadow -Werror -pedantic-errors -MD -MT modules/openthread/build/src/lib/spinel/CMakeFiles/openthread-spinel-rcp.dir/spinel.c.obj -MF modules/openthread/build/src/lib/spinel/CMakeFiles/openthread-spinel-rcp.dir/spinel.c.obj.d -o modules/openthread/build/src/lib/spinel/CMakeFiles/openthread-spinel-rcp.dir/spinel.c.obj   -c /projects/github/ncs/modules/lib/openthread/src/lib/spinel/spinel.c
In file included from /projects/github/ncs/zephyr/lib/libc/minimal/include/errno.h:20:0,
                 from /projects/github/ncs/modules/lib/openthread/src/lib/spinel/spinel.c:48:
/projects/github/ncs/zephyr/include/sys/errno_private.h:34:10: fatal error: syscalls/errno_private.h: No such file or directory
 #include <syscalls/errno_private.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```
so the dependency issue is introduced in that PR.

markus-becker-tridonic-com, 2020-09-21:
Where are the dependencies of OpenThread to Zephyr specified? Or how do you make a dependency from OpenThread onto the generation of `syscalls/ettno_private.h`?

tejlmand, 2020-09-21:
See comment here: https://github.com/zephyrproject-rtos/zephyr/pull/28458#issuecomment-696007802

The manifest file in Zephyr speficies the Openthread revision to use.

tejlmand, 2020-09-21:
closed, no bug found in Zephyr.
Invalid mixing of zephyr and openthread revisions.

markus-becker-tridonic-com, 2020-09-21:
I don't think this is solved. 
See my replies at https://github.com/zephyrproject-rtos/zephyr/pull/28458#issuecomment-696010538 and  https://github.com/zephyrproject-rtos/zephyr/pull/28458#issuecomment-696020300.

tejlmand, 2020-09-21:
Only happens when building with `-DCONFIG_MINIMAL_LIBC=y`.

PR #28562 (2020-09-24): west.yml: update openthread module
Fixes: #28465

Update openthread to dependency changes for fixing #28465.

see zephyrproject-rtos/openthread/pull/31

--------

Depends on: https://github.com/zephyrproject-rtos/openthread/pull/31
commit b88b1e66 (2020-09-24): west.yml: update openthread module

Fixes: #28465

Update openthread to dependency changes for fixing #28465.

see zephyrproject-rtos/openthread/pull/31
closed 2020-09-24 by carlescufi