zephyrproject-rtos/zephyr #28471: Central not working properly on nRF5340-DK

rick1082, 2020-09-17:
**Describe the bug**
When using nRF5340-DK as central, the central cannot establish connection with peripheral properly, disconnected happen immediately with reason 0x3e.  

**To Reproduce**
Steps to reproduce the behavior:
1. Program the "zephyr\samples\bluetooth\hci_rpmsg" on nRF5340-DK/nRF5340-PDK network core
2. Program the "zephyr\tests\bluetooth\shell" on nRF5340-DK/nRF5340-PDK application core 
3. Try to scan and connect with a nearby peripheral device 
4. The central site will not able to establish the connection

**Expected behavior**
The connection should be established properly, without disconnected immediately.

**Impact**
Not able to establish BLE link.

**Logs and console output**
uart:~$ bt connect e9:f5:61:22:63:5a random
Connection pending
Connected: e9:f5:61:22:63:5a (random)
Disconnected: e9:f5:61:22:63:5a (random) (reason 0x3e)

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: gnuarmemd
 - Commit SHA 2164a7d0dc51f727eeb90af346c6d35c96722ad8

**Additional context**
I tried to add the following configuration for excluding the issue from external clock source:
CONFIG_CLOCK_CONTROL_NRF_K32SRC_RC=y
CONFIG_CLOCK_CONTROL_NRF_K32SRC_250PPM=y
But no help :(

cvinayak, 2020-09-18:
@rick1082 Please try this: https://github.com/zephyrproject-rtos/zephyr/pull/27069

rick1082, 2020-09-18:
Thank you so much Vinayak :) It works.

ioannisg, 2020-09-21:
@rick1082  why have you closed this issue? Is your problem fixed, even though #27069 is not merged?

PR #27069 (2020-09-24): Bluetooth: controller: nRF53x: Fix missing sw_switch clear DPPI config
Fix missing sw_switch timer clear DPPI config when
re-enabling Tx or Rx after radio_disable() or
radio_switch_complete_and_disable() call in LLL state/role
contexts.

Fixes #28471.
commit 1c365951 (2020-09-24): Bluetooth: controller: nRF53x: Fix missing sw_switch clear DPPI config

Fix missing sw_switch timer clear DPPI config when
re-enabling Tx or Rx after radio_disable() or
radio_switch_complete_and_disable() call in LLL state/role
contexts.

Fixes #28471.
closed 2020-09-24 by ioannisg