zephyrproject-rtos/zephyr #28492: Could not build Zephyr application for swervolf_nexys board in simulation

cst-kirank, 2020-09-18:
I am getting an error while building Zephyr application for swervolf_nexys board in simulation.
The error states that "Unsupported board: led0 devicetree alias is not defined".

According to this error, the swervolf_nexys board is not supported by Zephyr application but the same is instructed in the README file of SweRVolf corers.

These are the few more details for the reference.
- I am using Linux Ubuntu distribution version 16.04.
- I have installed zephyr-sdk 0.11.3 application in $HOME/zephyr-sdk-0.11.3/ directory with zephyrproject in $HOME/zephyrproject/ directory.
- I have been able to run the pre-compiled zephyr application successfully as it was instructed in README file of SweRVolf.

Following is the command prompt output for the reference,

kirank@CSTLinux1:~/zephyrproject/zephyr/samples/basic/blinky$ west build -b swervolf_nexys -- -DBOARD_ROOT=$SWERVOLF_ROOT/zephyr -DSOC_ROOT=$SWERVOLF_ROOT/zephyr -DDTS_ROOT=$SWERVOLF_ROOT/zephyr
-- west build: generating a build system
Including boilerplate (Zephyr base (cached)): /home/kirank/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/kirank/zephyrproject/zephyr/samples/basic/blinky
-- Zephyr version: 2.4.0-rc1 (/home/kirank/zephyrproject/zephyr)
-- Found west (found suitable version "0.7.3", minimum required is "0.7.1")
-- Board: swervolf_nexys
-- Cache files will be written to: /home/kirank/.cache/zephyr
-- Found toolchain: zephyr (/home/kirank/zephyr-sdk-0.11.3)
-- Found dtc: /home/kirank/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/kirank/swervolf/fusesoc_libraries/swervolf/zephyr/boards/riscv/swervolf_nexys/swervolf_nexys.dts
-- Generated zephyr.dts: /home/kirank/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/kirank/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/include/generated/devicetree_unfixed.h
Parsing /home/kirank/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/kirank/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/.config'
No change to configuration in '/home/kirank/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/.config'
No change to Kconfig header in '/home/kirank/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/include/generated/autoconf.h'
CMake Warning at ../../../CMakeLists.txt:408 (message):
One or more dts_fixup.h files detected:

/home/kirank/swervolf/fusesoc_libraries/swervolf//zephyr/soc/riscv/swervolf/dts_fixup.h

Use of these files is deprecated; use the devicetree.h API instead.

-- Configuring done
-- Generating done
-- Build files have been written to: /home/kirank/zephyrproject/zephyr/samples/basic/blinky/build
-- west build: building application
[10/92] Building C object CMakeFiles/app.dir/src/main.c.obj
FAILED: ccache /home/kirank/zephyr-sdk-0.11.3/riscv64-zephyr-elf/bin/riscv64-zephyr-elf-gcc -DBUILD_VERSION=v2.4.0-rc1-197-g2164a7d0dc51 -DKERNEL -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -I/home/kirank/zephyrproject/zephyr/include -Izephyr/include/generated -I/home/kirank/swervolf/fusesoc_libraries/swervolf/zephyr/soc/riscv/swervolf -isystem /home/kirank/zephyrproject/zephyr/lib/libc/minimal/include -isystem /home/kirank/zephyr-sdk-0.11.3/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/9.2.0/include -isystem /home/kirank/zephyr-sdk-0.11.3/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/9.2.0/include-fixed -Os -imacros /home/kirank/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mabi=ilp32 -march=rv32ima -imacros /home/kirank/zephyrproject/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wno-address-of-packed-member -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/kirank/zephyrproject/zephyr/samples/basic/blinky=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/kirank/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/kirank/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -MD -MT CMakeFiles/app.dir/src/main.c.obj -MF CMakeFiles/app.dir/src/main.c.obj.d -o CMakeFiles/app.dir/src/main.c.obj -c ../src/main.c
../src/main.c:24:2: error: #error "Unsupported board: led0 devicetree alias is not defined"
24 | #error "Unsupported board: led0 devicetree alias is not defined"
| ^~~~~
[10/92] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/onoff.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake --build

I do not understand the cause of the above error and it will be great to hear from you.

Thanks,
Kiran Khunte

galak, 2020-09-18:
It looks like the SweRVolf support is not zephyr upstream, so this question might be better asked to wherever you got the SweRVolf support.

carlescufi, 2020-09-18:
This IC is supported in this fork: https://github.com/olofk/zephyr/commits/swervolf

I suggest you contact @olofk for more info.

closed 2020-09-18 by carlescufi

cst-kirank, 2020-09-19:
Yes,
Thank you for the suggestion and link.