zephyrproject-rtos/zephyr #28499: Solution for increased power consumption after enabling GPIO interrupt on nRF52832

anangl, 2020-09-18:
It can be observed on the nRF52832 SoC that just enabling an edge triggered GPIO interrupt (namely, calling
`gpio_pin_interrupt_configure()` or `gpio_pin_configure()` with one of the `GPIO_INT_EDGE_*` flags) leads to increasing the power consumption by ~15 uA. This happens because currently (in Zephyr v2.4.0-rc1) the GPIO driver for nRF SoCs uses GPIOTE channels to detect transitions on GPIO lines and to generate the edge triggered interrupts (while the GPIO PORT event is used only for the level triggered interrupts), and the nRF52832 SoC is affected by [anomaly 97](https://infocenter.nordicsemi.com/topic/errata_nRF52832_Rev2/ERR/nRF52832/Rev2/latest/anomaly_832_97.html).

While it is possible (although with less accuracy) to detect transitions on GPIO lines using the PORT event (and this is the workaround provided in the mentioned above anomaly description), the Zephyr GPIO driver does not provide such possibility. It would be nice to enhance the driver with such option and to provide some way for the users to select the detection method (even for nRF52832 it could be still desirable to use a GPIOTE channel for this).

Until the driver is improved, one of the following approaches can be used as a solution for the problem:
- use level triggered interrupts instead, i.e. use `GPIO_INT_LEVEL_*` flags to configure the interrupt (this requires also reconfiguration of the interrupt in its callback, otherwise the callback will be called again as long as the configured level is still present on the line)
- use the nrfx GPIOTE driver instead of the Zephyr GPIO one (it would need to be disabled with `CONFIG_GPIO=n` then)
  see the [`samples/boards/nrf/nrfx` sample](https://github.com/zephyrproject-rtos/zephyr/tree/17c4396ed42bda13a7a5f2c2edb55528a008fe0e/samples/boards/nrf/nrfx) for a reference how to do it (the DPPI related part should be omitted in this context); the crucial thing is to select the PORT event as the edge detection method by applying
  ```
  		.hi_accuracy = false,
  ```
  in the input configuration:
  https://github.com/zephyrproject-rtos/zephyr/blob/17c4396ed42bda13a7a5f2c2edb55528a008fe0e/samples/boards/nrf/nrfx/src/main.c#L43-L49

---

A related ticket in the Nordic Semiconductor's DevZone be found [here](https://devzone.nordicsemi.com/f/nordic-q-a/66002/zephyr-ncs---gpio_pin_interrupt_configure-causes-15ua-increase-in-consumption).

mhx, 2020-10-22:
FYI, I just found this issue after running into exactly the same problem with an nRF52840, and switching from edge triggered to level triggered interrupts also solved it for me. This appears to be an "Engineering A" sample (NRF_FICR->INFO.VARIANT reads AAC0, the chip is actually under a metal can on a module), and according to the errata this also suffers from anomaly 97.

mhx, 2020-10-23:
Ah, no, AAC0 should actually be indicative of a "Revision 1" variant, which should have anomaly 97 fixed. So I'm not sure why I'm seeing this problem.

mhx, 2020-10-30:
I've also tried a Revision 2 chip now (`INFO.VARIANT` == AAD0), which shows the same issue.

trembel, 2022-01-21:
Experiencing the same issue.. When will this be fixed?

sigurdnev, 2022-01-21:
> Experiencing the same issue.. When will this be fixed?

Try setting CONFIG_GPIO_NRF_INT_EDGE_USING_SENSE=y

anangl, 2022-01-21:
> > Experiencing the same issue.. When will this be fixed?
> 
> Try setting CONFIG_GPIO_NRF_INT_EDGE_USING_SENSE=y

This option is no longer available. It has been replaced by the [sense-edge-mask property in "nordic,nrf-gpio" compatible nodes](https://github.com/zephyrproject-rtos/zephyr/blob/cbd93a819854c3c8c7410ac2ab99fd42108b2630/dts/bindings/gpio/nordic%2Cnrf-gpio.yaml#L20-L25).

anangl, 2022-01-21:
Addressed in #31823 and #41288.

closed 2022-01-21 by anangl