zephyrproject-rtos/zephyr #28504: dfu: dfu libraries might fails to compile on redefined  flash_area_erased_val() while building MCUBoot

nvlsianpu, 2020-09-18:
**Describe the bug**
flash_area_erased_val() function is provide either by mcuboot and dfu/boot/mcuboot.c

**To Reproduce**
build MCUBoot with `BOOT_WAIT_FOR_USB_DFU=y`

**Expected behavior**
build succeed

**Impact**
Can't use USB DFU class in mcuboot

**Additional context**
https://github.com/zephyrproject-rtos/zephyr/commit/17c4396ed42bda13a7a5f2c2edb55528a008fe0e

d3zd3z, 2020-09-22:
I consider that `flash_area_erased_val()` should be removed from MCUboot, as it was provided due to being missing in Zephyr. Now that it is available in Zephyr, the workaround in MCUboot can be removed.

utzig, 2020-09-22:
> I consider that `flash_area_erased_val()` should be removed from MCUboot, as it was provided due to being missing in Zephyr. Now that it is available in Zephyr, the workaround in MCUboot can be removed.

https://github.com/JuulLabs-OSS/mcuboot/pull/808

nvlsianpu, 2020-09-23:
I have been thinking about making a module in mcuboot (around bootutil_misc.c file) for rest of redefined functions, but this would be over-kill and would be non trivial to meet all of code dependency. Also will add zephyr from mcuboot dependency for application build (and dependency while building mcuboot would be circular).

d3zd3z, 2020-09-23:
@agansari Fixing this issue will allow us to have an implementation of this for lpc55s69 that uses the controller's query for erase. This should improve reliability of MCUboot on these devices.

agansari, 2020-09-25:
@d3zd3z this means refactoring `is_area_readable()` function, i.e. moving part of the code to `flash_area_read_is_empty()`.

closed 2020-11-13 by carlescufi