zephyrproject-rtos/zephyr #28535: RFC: Add lz4 Data Compresssion library Support

Navin-Sankar, 2020-09-19:
## Introduction
This proposal plans to add support for minimal lz4 data compression & decompression library as an external module.

### Problem description
Currently Zephyr doesn't have any data compression & decompression library support. Data compression & decompression library enables the efficiency in communication with cloud platforms, OTA and various other use cases by reducing the data bandwidth for resource constrainted and battery operated devices.

### Proposed change
lz4 library is forked and zephyr related changes are added here https://github.com/linumiz/lz4

## Detailed RFC
External lz4 library can reviewed from here: https://github.com/linumiz/lz4

## Proposed change (Detailed)
modules: lib: add lz4 library to support minimal lz4 fast compression and decompression.
samples: compression: lz4: sample application will compress and decompress the given data.

## Dependencies
NA

## Concerns and Unresolved Questions
NA

## Alternatives
NA

Navin-Sankar, 2020-11-19:
@carlescufi requesting for review.

carlescufi, 2020-12-07:
> @carlescufi requesting for review.

@Navin-Sankar as far as I can tell you have already followed all the steps in:
https://docs.zephyrproject.org/latest/guides/modules.html#process-for-submitting-a-new-module

@ioannisg @nashif @MaureenHelm @galak any objections to me creating the repo? I propose naming it simply `lz4`.

Navin-Sankar, 2021-02-08:
@carlescufi @ioannisg @nashif @MaureenHelm @galak requesting for review!

carlescufi, 2021-02-10:
> @carlescufi @ioannisg @nashif @MaureenHelm @galak requesting for review!

We will discuss this in the TSC meeting today at 17h CET.

carlescufi, 2021-02-10:
> > @carlescufi @ioannisg @nashif @MaureenHelm @galak requesting for review!
> 
> We will discuss this in the TSC meeting today at 17h CET.

The TSC has agreed to create the repository and add lz4 as a module.

carlescufi, 2021-02-18:
https://github.com/zephyrproject-rtos/lz4

closed 2021-02-18 by carlescufi