zephyrproject-rtos/zephyr #28544: Null pointer dereference in ll_adv_aux_ad_data_set

jcyrax, 2020-09-20:
**Describe the bug**
When updating advertising data with `bt_le_adv_update_data` I get error and warning message in log: "bt_hci_core: opcode 0x2037 status 0x0d". I'm not using extended advertising. 

I debugged the problem to be in `ll_adv_aux_ad_data_set` where controller updates advertising data and after that tries to start extended advertising even if it is not used, however it doesn't check `adv->lll.aux` for NULL.

**To Reproduce**
I'm running on custom board with modifications to zephyr, but I think the problem should be reproducible with any BT board as long as CONFIG_BT_EXT_ADV=y or even by looking at the source.

**Expected behavior**
`bt_le_adv_update_data` should return 0.

**Impact**
Fixed in my fork so no problem for me.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: gcc-arm-none-eabi-8-2018-q4-major
 - Commit 122bc38fad422226fc1fc62ef844afa035f0fefe + custom modifications

cvinayak, 2020-09-22:
@jcyrax Could you please share a simple sample to reproduce this issue, so that I am able to provide a fix... since you fixed in your fork, you could send a PR too.

Thank you for reporting,

-Vinayak

jcyrax, 2020-09-22:
I'll try to do that, but it will take at least couple of days.

cvinayak, 2020-10-01:
Updating legacy advertising data using extended advertising commands (experimental in controller) causes BUS fault. Hence, increasing the issue priority to medium. A PR has been submitted by the reporter.

[image]

PR #28831 (2020-10-05): Bluetooth: controller: Fix null pointer deference during adv data update
Added a check for adv->lll.aux before starting extended advertising to
fix observed NULL pointer dereference when updating AD data of legacy
advertising.

Fixes #28544.
commit ec9080a0 (2020-10-05): Bluetooth: controller: Fix Null deref during adv data update

Added a check for adv->lll.aux before starting extended advertising to
fix observed NULL pointer dereference when updating AD data of legacy
advertising.

Fixes #28544.
closed 2020-10-05 by carlescufi