zephyrproject-rtos/zephyr #28547: up_squared: tests/subsys/debug/coredump failed.

chen-png, 2020-09-21:
**Describe the bug**
failed to run testcase  "coredump.logging_backend" on up_squared platform

**To Reproduce**
Steps to reproduce the behavior:
1.  west build -b up_squared -p auto tests/subsys/debug/coredump/
2.  west flash ~/up_squared.sh
3.  see error

**Logs and console output**
*** Booting Zephyr OS build v2.4.0-rc2  ***
Coredump: up_squared
E: Page fault at address 0x0 (error code 0x2)
E: Linear address not present in page tables
E: Access violation: supervisor thread not allowed to write
E: PTE: not present
E: RAX: 0x0000000000104418 RBX: 0x0000000000000000 RCX: 0x0000000000000046 RDX: 0x0000000000000000
E: RSI: 0x000000000000000a RDI: 0x0000000000000000 RBP: 0x0000000000000000 RSP: 0x0000000000d0afd0
E:  R8: 0x0000000000000000  R9: 0x0000000000000001 R10: 0x0000000000000000 R11: 0x0000000000000000
E: R12: 0x0000000000000000 R13: 0x0000000000000000 R14: 0x0000000000000000 R15: 0x0000000000000000
E: RSP: 0x0000000000d0afd0 RFLAGS: 0x0000000000010247 CS: 0x0018 CR3: 0x0000000001117000
E: RIP: 0x0000000000100012
**E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x0000000000106620 (unknown)**
E: #CD:BEGIN#
E: #CD:5a4501000200060000000000
E: #CD:410100b000
E: #CD:0e00000000000000020000000000000018441000000000004600000000000000
E: #CD:00000000000000000a000000000000000000000000000000d0afd00000000000
E: #CD:0000000000000000010000000000000000000000000000000000000000000000
E: #CD:1200100000000000470201000000000018000000000000002000000000000000
E: #CD:0000000000000000000000000000000000000000000000000000000000000000
E: #CD:00000000000000000000000000000000
E: #CD:4d010000501000000000000080110100000000
E: #CD:0000000000000000000000000000000000000000000000000000000000000000
E: #CD:0000000000000000000000000000000000000000000000000000000000000000
E: #CD:2000000000000000210000000000000022000000000000002300000000000000
E: #CD:2400000000000000250000000000000026000000000000002700000000000000
E: #CD:280000000000000029000000000000002a000000000000002b00000000000000
E: #CD:2c000000000000002d000000000000002e000000000000002f00000000000000
E: #CD:3000000000000000310000000000000032000000000000003300000000000000
E: #CD:3400000000000000350000000000000036000000000000003700000000000000
E: #CD:380000000000000039000000000000003a000000000000003b00000000000000
E: #CD:3c000000000000003d000000000000003e000000000000003f00000000000000
E: #CD:4000000000000000410000000000000042000000000000004300000000000000

**Environment (please complete the following information):**
 - OS: Fedora31
 - Toolchain: zephyr-sdk-0.11.4
 - Commit ID: 122bc38fad

dcpleung, 2020-09-21:
This is not a bug. The coredump test actually causes an exception so coredump can be performed. Closing...

chen-png, 2021-01-13:
reopen this, it failed using twister

PR #31331 (2021-01-15): tests: debug/coredump: change to dump minimal memory
The default of coredump is to dump all RAM as defined in
the linker script. However, this takes a while on boards
with bigger memory space, and would slow down automated
testing due to the need to grab all the data via serial.
So change it to dump minimal memory, as this would still
test the memory dumping capability.

Fixes #28547
commit e9c28e41 (2021-01-15): tests: debug/coredump: change to dump minimal memory

The default of coredump is to dump all RAM as defined in
the linker script. However, this takes a while on boards
with bigger memory space, and would slow down automated
testing due to the need to grab all the data via serial.
So change it to dump minimal memory, as this would still
test the memory dumping capability.

Fixes #28547
closed 2021-01-15 by nashif