zephyrproject-rtos/zephyr #28552: up_squared: samples/portability/cmsis_rtos_v1/philosophers/ failed.

chen-png, 2020-09-21:
**To Reproduce**
Steps to reproduce the behavior:
1.  west build -b up_squared -p auto samples/portability/cmsis_rtos_v1/philosophers/
2. west flash ~/up_squared.sh
3. see error

**Logs and console output**
*** Booting Zephyr OS build v2.4.0-rc2  ***
Running test suite test_cmsis_v2_apis

START - test_kernel_apis
PASS - test_kernel_apis

START - test_delay
PASS - test_delay

**START - test_thread_apis
Assertion failed at WEST_TOPDIR/zephyr/tests/portability/cmsis_rtos_v2/src/thread_apis.c:131: thread_apis_common: (osThreadGetCount() not equal to 2)
Incorrect number of cmsis rtos v2 threads
FAIL - test_thread_apis**

START - test_thread_apis_dynamic
Assertion failed at WEST_TOPDIR/zephyr/tests/portability/cmsis_rtos_v2/src/thread_apis.c:82: thread2: (num_threads not equal to 2)
PASS - test_thread_apis_dynamic
Incorrect number of cmsis rtos v2 threads

START - test_thread_prio^M
PASS - test_thread_prio^

**Environment (please complete the following information):**
 - OS: Fedora31
 - Toolchain: zephyr-sdk-0.11.4
 - Commit ID: 122bc38fad

dcpleung, 2020-09-21:
This does not seem to be the correct log. Could you update?

chen-png, 2020-09-23:
> This does not seem to be the correct log. Could you update?

sorry, pasted error log. I have updated the log.

PR #28657 (2020-09-24): samples: cmsis_rtos_v1: philosophers: skip up_squared
The up_squared board suffers the same issue as qemu_x86_64
where a bigger stack is needed but CMSIS has a limit on
how big the stack can be. This results in stack overflow.

So exclude up_squared in samples.yaml.

Fixes #28552
commit e37f819d (2020-09-24): samples: cmsis_rtos_v1: philosophers: skip up_squared

The up_squared board suffers the same issue as qemu_x86_64
where a bigger stack is needed but CMSIS has a limit on
how big the stack can be. This results in stack overflow.

So exclude up_squared in samples.yaml.

Fixes #28552
closed 2020-09-24 by MaureenHelm