zephyrproject-rtos/zephyr #28556: mec15xxevb_assy6853:running tests/kernel/sched/schedule_api/ failed

Zhaoningx, 2020-09-21:
Describe the bug
running tests/kernel/sched/schedule_api/ error, it showed me assertion failed.

To Reproduce
Steps to reproduce the behavior:

sanitycheck --hardware-map ~/mec1.file  --device-testing  -T tests/kernel/sched/schedule_api/
see error

START - test_slice_reset
PASS - test_slice_reset

===================================================================
START - test_slice_scheduling
AB
AB
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
A
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
FAIL - test_slice_scheduling

==================================================================
START - test_priority_scheduling
AB
AB
AB
AB
AB
PASS - test_priority_scheduling

===================================================================
START - test_wakeup_expired_timer_thread
PASS - test_wakeup_expired_timer_thread

===================================================================
START - test_user_k_wakeup
PASS - test_user_k_wakeup

===================================================================
START - test_user_k_is_preempt
PASS - test_user_k_is_preempt

===================================================================
Test suite threads_scheduling failed.

===================================================================
PROJECT EXECUTION FAILED

Environment (please complete the following information):
OS: Fedora28
Toolchain: Zephyr-sdk-0.11.3
Commit ID: 122bc38fad

chrta, 2020-09-25:
The same test fails for me with the current master (6a15118937a817968701b386b3f3c4b47879bde3) on the efm32pg_stk3402a board (ubuntu 18.04, zephyr sdk 0.11.1):

```
$ sanitycheck -p efm32pg_stk3402a -v --device-testing --device-serial /dev/ttyACM1 -s tests/kernel/sched/schedule_api/kernel.scheduler

...

$ cat .../handler.log

...

START - test_slice_reset
PASS - test_slice_reset
===================================================================
START - test_slice_scheduling
ABCDEFGHI
AB
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
C
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
D
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
E
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
F
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
G
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
H
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
I
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
A
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
FAIL - test_slice_scheduling
===================================================================
START - test_priority_scheduling
ABCDEFGHI
ABCDEFGHI
ABCDEFGHI
ABCDEFGHI
ABCDEFGHI
PASS - test_priority_scheduling
```
The test `tests/kernel/sched/schedule_api/kernel.scheduler.multiq` fails in the same way.

maksimmasalski, 2020-10-13:
> The same test fails for me with the current master ([6a15118](https://github.com/zephyrproject-rtos/zephyr/commit/6a15118937a817968701b386b3f3c4b47879bde3)) on the efm32pg_stk3402a board (ubuntu 18.04, zephyr sdk 0.11.1):
> 
> ```
> $ sanitycheck -p efm32pg_stk3402a -v --device-testing --device-serial /dev/ttyACM1 -s tests/kernel/sched/schedule_api/kernel.scheduler
> 
> ...
> 
> $ cat .../handler.log
> 
> ...
> 
> START - test_slice_reset
> PASS - test_slice_reset
> ===================================================================
> START - test_slice_scheduling
> ABCDEFGHI
> AB
> Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
> C
> Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
> D
> Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
> E
> Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
> F
> Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
> G
> Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
> H
> Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
> I
> Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
> Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
> A
> Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
> FAIL - test_slice_scheduling
> ===================================================================
> START - test_priority_scheduling
> ABCDEFGHI
> ABCDEFGHI
> ABCDEFGHI
> ABCDEFGHI
> ABCDEFGHI
> PASS - test_priority_scheduling
> ```
> 
> The test `tests/kernel/sched/schedule_api/kernel.scheduler.multiq` fails in the same way.

Thanks for that data. At least, I see both boards are Cortex M4, so it can help me to investigate a root cause.

maksimmasalski, 2020-10-14:
I don't see that bug happens on the Microchip board using the latest commit 
`sanitycheck --hardware-map ~/mec1.file --device-testing -T tests/kernel/sched/schedule_api/ -vv`
34cf1c74c21b10acfbc6504cd9bc212317c726d0
For the efm32pg_stk3402a I don't have that hardware, could you check that test again @chrta ? If you still see the bug for your board, please open new issue. I will close that one.

maksimmasalski, 2020-10-14:
```
INFO    - JOBS: 8
DEBUG   - Reading test case configuration files under /home/ztest/work/zephyrpro                                                                                                                                                       ject/zephyr/tests/kernel/sched/schedule_api...
DEBUG   - scanning /home/ztest/work/zephyrproject/zephyr/tests/kernel/sched/sche                                                                                                                                                       dule_api
DEBUG   - Found possible test case in /home/ztest/work/zephyrproject/zephyr/test                                                                                                                                                       s/kernel/sched/schedule_api
DEBUG   - Reading platform configuration files under /home/ztest/work/zephyrproj                                                                                                                                                       ect/zephyr/boards...
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boa                                                                                                                                                       rds/arc/emsdp/emsdp_em7d_esp.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boa                                                                                                                                                       rds/arc/emsdp/emsdp_em4.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boa                                                                                                                                                       rds/arc/emsdp/emsdp_em5d.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boa                                                                                                                                                       rds/arc/emsdp/emsdp.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boa                                                                                                                                                       rds/arc/emsdp/emsdp_em7d.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boa                                                                                                                                                       rds/arc/emsdp/emsdp_em6.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boa                                                                                                                                                       rds/arc/emsdp/emsdp_em9d.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boa                                                                                                                                                       rds/arc/em_starterkit/em_starterkit_em7d.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boa                                                                                                                                                       rds/arc/em_starterkit/em_starterkit_em11d.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boa                                                                                                                                                       rds/arc/em_starterkit/em_starterkit.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boa                                                                                                                                                       rds/arc/em_starterkit/em_starterkit_em7d_v22.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boa                                                                                                                                                       rds/arc/hsdk/hsdk.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boa                                                                                                                                                       rds/arc/hsdk/hsdk_2cores.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boa                                                                                                                                                       rds/arc/iotdk/iotdk.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boa                                                                                                                                                       rds/arc/nsim/nsim_hs_smp.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boa                                                                                                                                                       rds/arc/nsim/nsim_em.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boa                                                                                                                                                     [... cut ...]     238.54 seconds
INFO    - In total 72 test cases were executed on 1 out of total 292 platforms (                                                                                                                                                       0.34%)
INFO    - 4 tests executed on platforms, 0 tests were only built.

Hardware distribution summary:

| Board               | ID       |   Counter |
|---------------------|----------|-----------|
| mec15xxevb_assy6853 | FT9FPJYP |         4 |
```

closed 2020-10-14 by maksimmasalski

chrta, 2020-10-15:
Just for reference: The tests are now all PASSED for efm32pg_stk3402a.

```
$ sanitycheck -p efm32pg_stk3402a -v --device-testing --device-serial /dev/ttyACM1 -T tests/kernel/sched/schedule_api/ -vv
...
DEBUG   - run status: passed
INFO    - 1/4 efm32pg_stk3402a          tests/kernel/sched/schedule_api/kernel.scheduler   PASSED (device 19.918s)
INFO    - 2/4 efm32pg_stk3402a          tests/kernel/sched/schedule_api/kernel.scheduler.multiq PASSED (device 19.930s)
INFO    - 3/4 efm32pg_stk3402a          tests/kernel/sched/schedule_api/kernel.scheduler.no_timeslicing PASSED (device 4.392s)
INFO    - 4/4 efm32pg_stk3402a          tests/kernel/sched/schedule_api/kernel.scheduler.multiq_no_timeslicing PASSED (device 4.691s)

DEBUG   - running footprint_reports
INFO    - 4 of 4 tests passed (100.00%), 0 failed, 0 skipped with 0 warnings in 81.13 seconds
INFO    - In total 72 test cases were executed on 1 out of total 292 platforms (0.34%)
INFO    - 4 tests executed on platforms, 0 tests were only built.
...
```

maksimmasalski, 2020-10-15:
> Just for reference: The tests are now (revision [3ce7afd](https://github.com/zephyrproject-rtos/zephyr/commit/3ce7afdaaa005dc0f46d721ad77d0c275ce0df9f)) all PASSED for efm32pg_stk3402a.
> 
> ```
> $ sanitycheck -p efm32pg_stk3402a -v --device-testing --device-serial /dev/ttyACM1 -T tests/kernel/sched/schedule_api/ -vv
> ...
> DEBUG   - run status: passed
> INFO    - 1/4 efm32pg_stk3402a          tests/kernel/sched/schedule_api/kernel.scheduler   PASSED (device 19.918s)
> INFO    - 2/4 efm32pg_stk3402a          tests/kernel/sched/schedule_api/kernel.scheduler.multiq PASSED (device 19.930s)
> INFO    - 3/4 efm32pg_stk3402a          tests/kernel/sched/schedule_api/kernel.scheduler.no_timeslicing PASSED (device 4.392s)
> INFO    - 4/4 efm32pg_stk3402a          tests/kernel/sched/schedule_api/kernel.scheduler.multiq_no_timeslicing PASSED (device 4.691s)
> 
> DEBUG   - running footprint_reports
> INFO    - 4 of 4 tests passed (100.00%), 0 failed, 0 skipped with 0 warnings in 81.13 seconds
> INFO    - In total 72 test cases were executed on 1 out of total 292 platforms (0.34%)
> INFO    - 4 tests executed on platforms, 0 tests were only built.
> ...
> ```

Great!