zephyrproject-rtos/zephyr #28566: List property label for all active ADCs regardless of the board used

jakub-uC, 2020-09-21:
My proposal is related to issue #27350.

To solve this problem I want to modify the shell commands so that only active ADC channels can be selected. In other words, I will add subcommands with the names (property label) of active ADC channels. Right now it is hardcoded `ADC_0` and it is not compatible with all our boards supported by the ADC driver example.
In order to fix the issue, I would need an array with pointers to strings in "property: label" of each active ADC node (status "okay").

Unfortunately, I do not find an easy solution to my problem right now. I do not see any common compatible value or node name that will cover all ADC definitions in our DTSI files.

Nodes can have a name like: `adc`, `adc0`, or `adc1`.
Property labels that I found were: `ADC_0` or `ADC_1`
Property compatible - each vendor has its own compatible definition.

I have two solutions in my mind but all of them require some update on either all ADC drivers or all CPU's DTSI files.

First solution:
We can add one more compatible to the Device Tree: `adc` for all defined ADCs, for all CPUs.

Example change in existing DTSI for nrf52840:
```
adc: adc@40007000 {                                                                                      
                        compatible = "nordic,nrf-saadc";                                                          
                        reg = <0x40007000 0x1000>;                                                                       
                        interrupts = <7 1>;                                                                              
                        status = "disabled";                                                                             
                        label = "ADC_0";                                                                                 
                        #io-channel-cells = <1>;                                                                         
                }; 
```
change to:
```
adc: adc@40007000 {                                                                                      
                        compatible = "nordic,nrf-saadc", "adc";                                                          
                        reg = <0x40007000 0x1000>;                                                                       
                        interrupts = <7 1>;                                                                              
                        status = "disabled";                                                                             
                        label = "ADC_0";                                                                                 
                        #io-channel-cells = <1>;                                                                         
                }; 
```
Next, using existing macros in `include/devicetree.h` file I can look for all active nodes compatible with `adc`.

Second solution:
It is a proposal made by @erwango.
We can use the existing structure of the Device Tree and macros to create an array with pointers to all property label of all enabled ADCs:

```
# Need an intermediate macro to allow building a comma-separated list
#define NODE_LABELS(n) DT_INST_LABEL(n),

const struct flash_area labels_array[] = {
	DT_INST_FOREACH_STATUS_OKAY(NODE_LABELS)
};
```

But his part of the code needs to be located in each ADC driver .c file because only there we can find `DT_DRV_COMPAT` macro.
Next, I would need to reference this variable from the `adc_shell.c` file.

If there are any better solutions in place please let me know.

erwango, 2020-09-21:
@jakub-uC 

> But his part of the code needs to be located in each ADC driver .c file because only there we can find DT_DRV_COMPAT macro.
Next, I would need to reference this variable from the adc_shell.c file.

What about the existing code in `drivers/adc/adc_shell.c` :
```
#if DT_HAS_COMPAT_STATUS_OKAY(atmel_sam_afec)
#define DT_DRV_COMPAT atmel_sam_afec
#elif DT_HAS_COMPAT_STATUS_OKAY(atmel_sam0_adc)
#define DT_DRV_COMPAT atmel_sam0_adc
#elif DT_HAS_COMPAT_STATUS_OKAY(microchip_xec_adc)
#define DT_DRV_COMPAT microchip_xec_adc
#elif DT_HAS_COMPAT_STATUS_OKAY(nordic_nrf_adc)
#define DT_DRV_COMPAT nordic_nrf_adc
#elif DT_HAS_COMPAT_STATUS_OKAY(nordic_nrf_saadc)
#define DT_DRV_COMPAT nordic_nrf_saadc
#elif DT_HAS_COMPAT_STATUS_OKAY(nxp_kinetis_adc12)
#define DT_DRV_COMPAT nxp_kinetis_adc12
#elif DT_HAS_COMPAT_STATUS_OKAY(nxp_kinetis_adc16)
#define DT_DRV_COMPAT nxp_kinetis_adc16
#elif DT_HAS_COMPAT_STATUS_OKAY(st_stm32_adc)
#define DT_DRV_COMPAT st_stm32_adc
#else
#error No known devicetree compatible match for ADC shell
#endif
```

mbolivar-nordic, 2020-09-23:
> What about the existing code in `drivers/adc/adc_shell.c` ?

I'm not the biggest fan of this code, to be honest. It doesn't work well with out of tree drivers. They will need to edit the zephyr source code to make the shell aware of their device, for example, which is not so nice.

I kind of wonder if solution number 1 is acceptable to @galak .

erwango, 2020-09-24:
> I kind of wonder if solution number 1 is acceptable to @galak .

Would have to be something like "zephyr,adc-shell", then.
Even if working, this is yet another way to bind tests and board after "alias", "chosen" and "alternate label".
So we should be happy having lot of solutions ahead of us, but my concern is that this creates confusion in the end.
We absolutely need to give recommendations and clarify the uses.

mbolivar-nordic, 2020-09-24:
>  Even if working, this is yet another way to bind tests and board after "alias", "chosen" and "alternate label".

We've discussed alias is currently just for samples, and . Chosen seems wrong; we're not choosing anything, we want a list of available choices. Node labels won't work for the reason @jakub-uC pointed out in the description: different SoC vendors pick different ones and there is no consistency.

> Would have to be something like "zephyr,adc-shell", then.

I would have suggested simply "adc-controller" with no "zephyr," prefix. Is there some reason this would need a "zephyr" prefix?

martinjaeger, 2020-09-30:
This would also be helpful for the ADC sample, not only the ADC shell. See PR #25590 where we also started discussing how to handle the different adc names in the devicetree.

So if we go for solution 1, the compatible name should be generic and not contain `shell`, I think.

galak, 2020-09-30:
Is there a reason adc_shell doesn't do what i2c_shell does and use `shell_device_lookup`?

jakub-uC, 2020-09-30:
@galak If I will use `shell_device_lookup` please let me know how can I identify which device is ADC in a source code common for all boards?

galak, 2020-09-30:
So the idea of a generic compatible is not something I'm ok with.   I had the idea of a subsystem register (See PR #24885).  We should figure out what we actually need to support for the shell and samples.  Is it wanting a list of device "names" for a given driver subystem?

galak, 2020-09-30:
> @galak If I will use `shell_device_lookup` please let me know how can I identify which device is ADC in a source code common for all boards?

Not sure I follow the question here -- can you clarify.

jakub-uC, 2020-09-30:
In the example, I am trying to fix adc_shell, I need to list all ADCs available for the currently used board. Function `shell_device_lookup` lists all active devices. How I can determine which device is an ADC?
As I understand the problem for one board this function will give me the device name `ADC_0` for other `ADC_1`.

galak, 2020-09-30:
> In the example, I am trying to fix: adc_shell.c, I need to list all ADCs available for the currently used board. Function `shell_device_lookup` lists all active devices. How I can determine which device is an ADC?
> As I understand the problem for one board this function will give me the device name `ADC_0` for other `ADC_1`.

Ah, I think right now the best we do is assume ADC prefix in the name.  (see `I2C_DEVICE_PREFIX` in `drivers/i2c/i2c_shell.c`).

erwango, 2020-09-30:
Is there any way to get that info on the dts binding side ?
Maybe we can infer information based on the fact that each adc vendor compat include "adc-controller" common binding?

galak, 2020-09-30:
> Is there any way to get that info on the dts binding side ?
> Maybe we can infer information based on the fact that each adc vendor compat include "adc-controller" common binding?

I'd rather not make such assumptions, plus we don't track the include info at that level.

Somewhere we have to encode a DTS compatible to Zephyr subsystem mapping.  I see we can do this in a few ways:
1. Add some property like 'zephyr,subsystem' to dts: `zephyr,subsystem = "adc";`
2. encode mapping in build system (CMakeLists.txt: `zephyr_library_sources_dt("adc", "st,stm32-adc", adc_stm32.c)`)
3. encode as part of a variant of `DEVICE_AND_API_INIT()`: `ADC_DEVICE_AND_API_INIT(...)` (can build a table of device pointers that match a subsystem)

mbolivar-nordic, 2020-09-30:
> DTS compatible to Zephyr subsystem mapping

I might phrase this more generally as "device <--> API mapping" since not all devices come from DT, but I agree.

>     3\. encode as part of a variant of `DEVICE_AND_API_INIT()`: `ADC_DEVICE_AND_API_INIT(...)` (can build a table of device pointers that match a subsystem)

So something like a `CONFIG_ADC_DEVICE_TABLE` that's `default n`? And if it's set to `y`, `ADC_DEVICE_AND_API_INIT()` would stick the device into the table?

If so, that would let the `ADC_SHELL` option `imply ADC_DEVICE_TABLE` and things could just work, right?

galak, 2020-09-30:
> I might phrase this more generally as "device <--> API mapping" since not all devices come from DT, but I agree.
>
>
> So something like a `CONFIG_ADC_DEVICE_TABLE` that's `default n`? And if it's set [...]

Yes, that's inline with my thinking.

mbolivar-nordic, 2020-09-30:
> > If so, that would let the `ADC_SHELL` option `imply ADC_DEVICE_TABLE` and things could just work, right?
> 
> Yes, that's inline with my thinking.

:+1:

pabigot, 2020-10-01:
> Somewhere we have to encode a DTS compatible to Zephyr subsystem mapping. I see we can do this in a few ways:

Is that what we really need to do?  Don't we simply need to identify the available devices that implement a particular subsystem API?

If userspace is enabled the infrastructure to do this is already in place, because the kernel object associated with the device knows what type of device it is.  If userspace is not enabled I don't see an existing solution, but I don't think we need to get DTS compatibles involved here.  It should be sufficient to add a type field to `struct device` that gets set by `DEVICE_DEFINE()`.

galak, 2020-10-01:
Dev-Review (Oct 1 2020): Idea is to utilize the functionality that @pabigot describes that exists for userspace with K_OBJ_DRIVER_<foo> and see what the PR looks like.

pabigot, 2020-10-08:
@jakub-uC I'm willing to submit an implementation along the lines we discussed, but don't want to jump in if you've got it handled.

jakub-uC, 2020-10-09:
@pabigot : Please go ahead, I am currently working on something else.

erwango, 2024-02-13:
Looks like this should have been closed following merge of https://github.com/zephyrproject-rtos/zephyr/pull/28599.
Feel free to re-open of required

closed 2024-02-13 by erwango