zephyrproject-rtos/zephyr #28589: up_squared_adsp:running tests/kernel/workq/work_queue/ failed

KangJianX, 2020-09-22:
Describe the bug
running test/kernel/workq/work_queue/ error, it showed assertion failed.

To Reproduce
Steps to reproduce the behavior:

sanitycheck -W -p up_squared_adsp  --device-testing -T tests/kernel/workq/work_queue/ --west-flash="/home/ztest/work/zephyrproject/zephyr/boards/xtensa/up_squared_adsp/tools/up_squared_adsp_flash.sh, /home/ztest/work/sof/rimage/keys/otc_private_key.pem" --device-serial-pty="/home/ztest/work/ambient/set_ambient /home/ztest/work/zephyrproject/zephyr/boards/xtensa/up_squared_adsp/tools/mbterm.py" -x=CONFIG_IPM=y -x=CONFIG_CONSOLE=y -x=CONFIG_LOG_PRINTK=n

see error

PASS - test_triggered_wait

===================================================================
START - test_triggered_wait_expired
Starting triggered test
- Initializing triggered test items
- Submitting triggered test items
- Submitting triggered work 1
- Submitting triggered work 2
- Submitting triggered work 3
- Submitting triggered work 4
- Submitting triggered work 5
- Submitting triggered work 6
- Checking results (before timeout)
- Running triggered test item 1
- Running triggered test item 2
- Checking results (after timeout)
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/workq/work_queue/src/main.c:133: check_results: (num_results not equal t
o num_tests)
*** work items finished: 2 (expected: 6)
- Running triggered test item 3
- Running triggered test item 4
- Running triggered test item 5
- Running triggered test item 6
FAIL - test_triggered_wait_expired
===================================================================
Test suite workqueue failed.
===================================================================
PROJECT EXECUTION FAILED

Environment (please complete the following information):
OS: Fedora28
Toolchain: Zephyr-sdk-0.11.3
Commit ID: 2cd556a689

dcpleung, 2020-09-22:
Could you re-test? I could not reproduce it on master and on the commit you provided.

KangJianX, 2020-09-23:
Yes, I reproduce issue on this commit.

PR #28655 (2020-10-05): tests: work_queue: wait longer for expired triggering items
The test_triggered_wait_expired test submits the items with
2*SUBMIT_WAIT timeout and waits for the timeout to expire
so the items are being worked on. It waits one SUBMIT_WAIT
and checks none of the items have started. Then waits
another SUBMIT_WAIT to check if they have all finished.
However, since the timeout is at 2*SUBMIT_WAIT, the work
queue may have just started going through the list of items.
This means some items may have started while others have not.
This results in the test failing as not all items have
finished. So lengt [... cut]
commit 1cd545d5 (2020-10-05): tests: work_queue: wait longer for expired triggering items

The test_triggered_wait_expired test submits the items with
2*SUBMIT_WAIT timeout and waits for the timeout to expire
so the items are being worked on. It waits one SUBMIT_WAIT
and checks none of the items have started. Then waits
another SUBMIT_WAIT to check if they have all finished.
However, since the timeout is at 2*SUBMIT_WAIT, the work
queue may have just started going through the list of items.
This means some items may have started while others have not.
This results in the test failing as not all items have
finished. So leng [... cut]
closed 2020-10-05 by MaureenHelm