zephyrproject-rtos/zephyr #28590: up_squared_adsp:running tests/kernel/smp/ failed

KangJianX, 2020-09-22:
Describe the bug
running tests/kernel/smp/ error, it showed assertion failed.

To Reproduce
Steps to reproduce the behavior:

sanitycheck -W -p up_squared_adsp --device-testing -T tests/kernel/smp/ --west-flash="/home/ztest/work/zephyrproject/zephyr/boards/xtensa/up_squared_adsp/tools/up_squared_adsp_flash.sh, /home/ztest/work/sof/rimage/keys/otc_private_key.pem" --device-serial-pty="/home/ztest/work/ambient/set_ambient /home/ztest/work/zephyrproject/zephyr/boards/xtensa/up_squared_adsp/tools/mbterm.py" -x=CONFIG_IPM=y -x=CONFIG_CONSOLE=y -x=CONFIG_LOG_PRINTK=n

see error

*** Booting Zephyr OS build v2.4.0-rc2-23-g2cd556a689e3  ***
ASSERTION FAIL [!z_smp_cpu_mobile()] @ WEST_TOPDIR/zephyr/kernel/include/kswap.h:198
@ WEST_TOPDIR/zephyr/lib/os/assert.c:45

Environment (please complete the following information):
OS: Fedora28
Toolchain: Zephyr-sdk-0.11.3
Commit ID: 2cd556a

dcpleung, 2020-09-25:
This will be fixed by #26483

dcpleung, 2020-10-21:
@KangJianX Could you re-test? The SOF PR has been merged which should fix this.

KangJianX, 2020-10-22:
I can't re-test, all test cases are skiped when I run (sanitycheck -W -p up_squared_adsp --device-testing -T tests/kernel/smp/ --west-flash="/home/ztest/work/zephyrproject/zephyr/boards/xtensa/up_squared_adsp/tools/up_squared_adsp_flash.sh, /home/ztest/work/sof/rimage/keys/otc_private_key.pem" --device-serial-pty="/home/ztest/work/ambient/set_ambient /home/ztest/work/zephyrproject/zephyr/boards/xtensa/up_squared_adsp/tools/mbterm.py" -x=CONFIG_IPM=y -x=CONFIG_CONSOLE=y -x=CONFIG_LOG_PRINTK=n) this cmd. I block by [https://github.com/zephyrproject-rtos/zephyr/issues/29430](url)

KangJianX, 2020-12-21:
Verify this issue, it is fixed.

closed 2020-12-21 by KangJianX