zephyrproject-rtos/zephyr #28602: TCP2:frdm_k64f  tests/net/tcp2 regression failure in RC2

hakehuang, 2020-09-22:
**Describe the bug**
bus fault after test_client_syn_resend

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=frdm_k64f ..
3. make
4. See error

**Expected behavior**
test PASS

**Impact**
TCP2 stack issue

**Logs and console output**

```
 @ *** Booting Zephyr OS build v2.4.0-rc1-282-g122bc38fad42  ***
Running test suite test_tcp_fn
===================================================================
START - test_presetup
 PASS - test_presetup
===================================================================
START - test_client_ipv4
 PASS - test_client_ipv4
===================================================================
START - test_client_ipv6
 PASS - test_client_ipv6
===================================================================
START - test_server_ipv4
 PASS - test_server_ipv4
===================================================================
START - test_server_with_options_ipv4
 PASS - test_server_with_options_ipv4
===================================================================
START - test_server_ipv6
 PASS - test_server_ipv6
===================================================================
START - test_client_syn_resend
 PASSE: ***** BUS FAULT *****
E:   Imprecise data bus error
E: r0/a1:  0x2000a0d8  r1/a2:  0x00000000  r2/a3:  0x00000000
E: r3/a4:  0x00000000 r12/ip:  0x00000004 r14/lr:  0x0000cd4d
E:  xpsr:  0x6100020f
E: Faulting instruction address (r15/pc): 0x0000c79a
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Fault during interrupt handling

E: Current thread: 0x20001a38 (main)
E: Halting system

```

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v2.4.0-rc2

hakehuang, 2020-09-22:
@MaureenHelm , @dleach02

hakehuang, 2020-09-22:
mimxrt1020_evk is Ok, but mimxrt1064 fails.

jukkar, 2020-09-22:
Just tried this and there is no crash if I enable TCP debugging, seems to be some timing issue.

jukkar, 2020-09-22:
The issue happens in line 1685 in tcp2.c:`net_tcp_connect()`
```
		if (k_sem_take(&conn->connect_sem, timeout) != 0 &&
```
Running the unit test with this gives a bit more information:
`cmake -B build -DBOARD=frdm_k64f  . -DCONFIG_LOG_MINIMAL=n`
```
START - test_client_syn_resend
 PASSASSERTION FAIL [z_spin_lock_valid(l)] @ WEST_TOPDIR/zephyr/include/spinlock.h:92
        Recursive spinlock 0x20003590
ASSERTION FAIL [z_spin_lock_valid(l)] @ WEST_TOPDIR/zephyr/include/spinlock.h:92
        Recursive spinlock 0x20003590*** Booting Zephyr OS build v2.4.0-rc2-33-gc54a511d26d0  ***
Running test suite test_tcp_fn
```

PR #28624 (2020-09-23): tests: net: tcp2: Fix crash in frdm_k64f
Lowering the connection timeout in order to avoid spin lock assert
in frdm_k64f device. This error was seen with this device

ASSERTION FAIL [z_spin_lock_valid(l)] @ zephyr/include/spinlock.h:92
        Recursive spinlock 0x20003590

Fixes #28602
commit 4a27246c (2020-09-23): tests: net: tcp2: Fix crash in frdm_k64f

Lowering the connection timeout in order to avoid spin lock assert
in frdm_k64f device. This error was seen with this device

ASSERTION FAIL [z_spin_lock_valid(l)] @ zephyr/include/spinlock.h:92
        Recursive spinlock 0x20003590

Fixes #28602
closed 2020-09-23 by MaureenHelm