zephyrproject-rtos/zephyr #28603: tests: kernel: timer: timer_api: Failed on nucleo_L073rz

ABOSTM, 2020-09-22:
**Describe the bug**
tests/kernel/timer/timer_api/ failed for board nucleo_l073rz
2 steps are failed: test_timer_k_define and test_timer_remaining.
Both are linked to timing issue.
Please note that CPU run at 32MHz only.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p auto -b nucleo_l073rz tests/kernel/timer/timer_api/
2. west flash
3.  See error

**Expected behavior**
test passed

**Logs and console output**
```
*** Booting Zephyr OS build v2.4.0-rc1-166-g8e0f6a5936fa  ***
Running test suite timer_api
===================================================================
START - test_time_conversions
 PASS - test_time_conversions
===================================================================
START - test_timer_duration_period
 PASS - test_timer_duration_period
===================================================================
START - test_timer_restart
 PASS - test_timer_restart
===================================================================
START - test_timer_period_0
 PASS - test_timer_period_0
===================================================================
START - test_timer_expirefn_null
 PASS - test_timer_expirefn_null
===================================================================
START - test_timer_periodicity
 PASS - test_timer_periodicity
===================================================================
START - test_timer_status_get
 PASS - test_timer_status_get
===================================================================
START - test_timer_status_get_anytime
 PASS - test_timer_status_get_anytime
===================================================================
START - test_timer_status_sync
 PASS - test_timer_status_sync
===================================================================
START - test_timer_k_define

    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_api/src/main.c:94: duration_expire: (interval >= 100) || (((10000 % 1000U) != 0) && (interval == 100 - 1)) is false

 FAIL - test_timer_k_define
===================================================================
START - test_timer_user_data
 PASS - test_timer_user_data
===================================================================
START - test_timer_remaining

    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_api/src/main.c:642: test_timer_remaining: (abs(delta_ticks) <= MAX(slew_ticks, 1U) is false)
tick/busy slew -2 larger than test threshold 0
 FAIL - test_timer_remaining
===================================================================
START - test_timeout_abs
 PASS - test_timeout_abs
===================================================================
Test suite timer_api failed.
===================================================================
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS:  Linux and Windows
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA: 8e0f6a5936faa7c0bd48d5cc7d245f85cab22ddc

ioannisg, 2020-09-23:
the test is failing on multiple nRF52x platforms as well.

ABOSTM, 2020-09-23:
> the test is failing on multiple nRF52x platforms as well.

Does the proposed fix #28606 solve also issue on nRF52x platforms ?

ioannisg, 2020-09-23:
I'll give it a try @ABOSTM

PR #28606 (2020-09-24): tests: kernel: timer: timer_api: tick align and reorder
tests: kernel: timer: timer_api: tick align and reorder

* Insert k_usleep(1) just before k_timer_start()
    to guaranty tick alignment for step "test_timer_k_define"

* Move init_timer_data() out of k_usleep() tick alignment.
   And compute rem_ticks just after busy_wait_ms() to avoid slew
    due to 'now' and 'rem_ms' computations.  With slow CPU 32MHz: -2 Ticks.

Fixes #28603
closed 2020-09-24 by MaureenHelm