zephyrproject-rtos/zephyr #28613: cannot set watchpoints on x86 with icount enabled

andrewboie, 2020-09-22:
**Describe the bug**

If I try to watch a memory address:

```
(gdb) p &z_idle_threads[0].arch.rsi
$1 = (uint64_t *) 0x116f20 <z_idle_threads+352>
(gdb) x/fx 0x116f20
0x116f20 <z_idle_threads+352>:  0x00000000
(gdb) watch *0x116f20
Hardware watchpoint 2: *0x116f20
(gdb) c
Continuing.
```

The emulator explodes:

```
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.x2apic [bit 21]
SeaBIOS (version rel-1.12.1-0-ga5cab58-dirty-20200625_115407-9426dddc0a1f-zephyr
)
Booting from ROM..qemu: fatal: Raised interrupt while not in I/O function
EAX=00000000 EBX=00009500 ECX=000000df EDX=00000511
ESI=00000000 EDI=00116f20 EBP=00000000 ESP=00006f08
EIP=00008031 EFL=00000046 [---Z-P-] CPL=0 II=0 A20=1 SMM=0 HLT=0
ES =0010 00000000 ffffffff 00cf9300 DPL=0 DS   [-WA]
CS =0008 00000000 ffffffff 00cf9a00 DPL=0 CS32 [-R-]
SS =0010 00000000 ffffffff 00cf9300 DPL=0 DS   [-WA]
DS =0010 00000000 ffffffff 00cf9300 DPL=0 DS   [-WA]
FS =0010 00000000 ffffffff 00cf9300 DPL=0 DS   [-WA]
GS =0010 00000000 ffffffff 00cf9300 DPL=0 DS   [-WA]
LDT=0000 00000000 0000ffff 00008200 DPL=0 LDT
TR =0000 00000000 0000ffff 00008b00 DPL=0 TSS32-busy
GDT=     0000c070 0000004f
IDT=     00009000 00000fff
CR0=00000011 CR2=00000000 CR3=00000000 CR4=00000000
DR0=0000000000000000 DR1=0000000000000000 DR2=0000000000000000 DR3=0000000000000000
DR6=00000000ffff0ff0 DR7=0000000000000400
CCS=004a8000 CCD=00950000 CCO=CLR
EFER=0000000000000000
FCW=037f FSW=0000 [ST=0] FTW=00 MXCSR=00001f80
FPR0=0000000000000000 0000 FPR1=0000000000000000 0000
FPR2=0000000000000000 0000 FPR3=0000000000000000 0000
FPR4=0000000000000000 0000 FPR5=0000000000000000 0000
FPR6=0000000000000000 0000 FPR7=0000000000000000 0000
XMM00=00000000000000000000000000000000 XMM01=00000000000000000000000000000000
XMM02=00000000000000000000000000000000 XMM03=00000000000000000000000000000000
XMM04=00000000000000000000000000000000 XMM05=00000000000000000000000000000000
XMM06=00000000000000000000000000000000 XMM07=00000000000000000000000000000000
```

This only happens with icount enabled. I have seen this on x86 32-bit, and on x86_64 with CONFIG_MP_NUM_CPUS=1 (which enables icount since we only have one CPU)

**To Reproduce**
Steps to reproduce the behavior:
1. Build some test application and set a watch to some symbol or memory address in RAM in GDB
2. Continue execution

**Expected behavior**
GDB watches the memory as expected.

**Impact**
Annoyance

galak, 2020-09-24:
Can you try w/qemu 5.1:

builds.zephyrproject.org/zephyrproject-rtos/sdk-ng/272/zephyr-sdk-0.11.4-pr-272-x86_64-linux-setup.run

If still an issue, this should be filed w/upstream qemu.

closed 2020-12-08 by github-actions[bot]

stsquad, 2021-12-09:
The next QEMU release (6.2) has a fix for the watchpoint with icount bug.