zephyrproject-rtos/zephyr #28625: tests: net: tcp2: llegal use of the EPSR

ABOSTM, 2020-09-23:
**Describe the bug**
 tests/net/tcp2/ is failed on nucleo_f429zi, nucleo_f746zg (all boards running this test on STMicroelectronics test bench)

This is a regression introduced by #28302 Misc TCP2 fixes
and more specifically
 SHA1: 80d04616de61f6f2601d7a3ea7ff09bf7506af74 net: tcp2: Honor TCP retry count limit from Kconfig file  (git bisect)

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p auto -b nucleo_f429zi tests/net/tcp2/
2. west flash
3. See error

**Expected behavior**
Test passed

**Logs and console output**
```
*** Booting Zephyr OS build v2.4.0-rc1-280-gb5c2ff93974f  ***
Running test suite test_tcp_fn
===================================================================
START - test_presetup
 PASS - test_presetup
===================================================================
START - test_client_ipv4
 PASS - test_client_ipv4
===================================================================
START - test_client_ipv6
 PASS - test_client_ipv6
===================================================================
START - test_server_ipv4
 PASS - test_server_ipv4
===================================================================
START - test_server_with_options_ipv4
 PASS - test_server_with_options_ipv4
===================================================================
START - test_server_ipv6
 PASS - test_server_ipv6
===================================================================
START - test_client_syn_resend
 PASSE: ***** USAGE FAULT *****
E:   Illegal use of the EPSR
E: r0/a1:  0x2000c6d8  r1/a2:  0x20000fa0  r2/a3:  0x20000fa0
E: r3/a4:  0x00000000 r12/ip:  0x00000004 r14/lr:  0x0800c39b
E:  xpsr:  0x6000000f
E: Faulting instruction address (r15/pc): 0x00000000
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Fault during interrupt handling

E: Current thread: 0x20000fa0 (main)
E: Halting system
```

**Environment (please complete the following information):**
 - OS:  Linux and Windows
 - Toolchain:Zephyr SDK
 - Commit SHA :0e49f5570ce5d0434d45f115c7391dd95be81b07

jukkar, 2020-09-23:
Can you try #28624 if it helps for stm32? I just fixed similar issue for frdm_k64f.

ABOSTM, 2020-09-23:
@jukkar , yes #28624 solves this this on both boards nucleo_f429zi, nucleo_f746zg

jukkar, 2020-09-23:
Duplicate of #28602

closed 2020-09-23 by jukkar