zephyrproject-rtos/zephyr #28629: tests: kernel: common: and common.misra are failing on nrf52840dk

ioannisg, 2020-09-23:
**Describe the bug**
kernel.common and kernel.common.misra 

**To Reproduce**
Build and run the tests on nrf52840dk_nrf52840 and observe failure

**Expected behavior**
Test shall pass.

**Impact**
Kernel test failing.

**Logs and console output**

**Environment (please complete the following information):**
 - OS: both Linux and Windows
 - Toolchain: both Zephyr SDK, and Gnu ARM Embedded
 - Commit SHA or Version used

ioannisg, 2020-09-23:
CC @pabigot

ioannisg, 2020-09-23:
Failures seem to be intermittent so for now I remove the blocking priority

nashif, 2020-09-24:
> Failures seem to be intermittent so for now I remove the blocking priority

but this still needs a priority... :)

PR #28632 (2020-09-29): tests: kernel/common: align time test with system tick
Reduce the error between the timer (which is tick-aligned) and busy_wait (which is not) by aligning both.

Fixes #28629
closed 2020-09-29 by ioannisg