zephyrproject-rtos/zephyr #28634: usb: inconsistent behavior of usb_dc_ep_configure implenetation in the drivers

jfischer-phytec-iot, 2020-09-23:
Inconsistent behavior of usb_dc_ep_configure (and usb_dc_ep_enable) implementation in the drivers.

The drivers behave differently when the API function usb_dc_ep_configure is called again for already configured endpoints. For example usb_dc_kinetis and usb_dc_mcux_ehci abort with the message -EALREADY, usb_dc_nrfx reconfigures endpoint. 

Note: Behavior is not defined by the USB driver API and API function usb_dc_ep_flush is not used and (therefore probably) not implemented by all drivers.

If an interface will be (re)configured (Set Interface) three scenarios are possible: endpoint should be configured for the first time, endpoint should be reconfigured (also with the same parameters), endpoint should be disabled. So it is possible that the host calls "Set Interface" with actual interface alternate index to set the class or function to a defined state (for example to drop IN endpoint buffer). This and reconfiguration of the endpoint is not possible on all platforms due to inconsistent behavior. A uniform behavior should be documented and implemented, at the latest with rework from the USB device stack it should be fixed.

emob-nordic, 2020-11-13:
I think the `usb_dc_ep_configure()` should actually reconfigure the endpoint. Prio do doing this all transfers for particular endpoint should be dropped and endpoint should buffer should be flushed. Aborting with `-EALREADY` is inproper behaviour because same endpoint could be used with different MPS for different alternate settings. Looking into the future for USB core rework same endpoint could be used for 2 different classes swapped in runtime meaning that reconfiguration shall not return immediately if the endpoint was previously enabled with some other configuration.

carlescufi, 2021-05-19:
Switched to enhancement since this is not a bug per-se, but rather something that will be addressed in the future once the driver interface changes.

jfischer-no, 2023-05-04:
Improved, but not fully fixed. It will no longer be pursued as we move to the new USB support.
See also
cf91f1873babf32c9f720e5ad670721c98b0534c
f8ac6b6d669ccf6209817d8ced5b712c128115c1
e326c5839938386b945ca213b8fafd0e48d4f9dd

closed 2023-05-04 by jfischer-no