zephyrproject-rtos/zephyr #28638: bq274xx sample unable to build

mejtejt, 2020-09-23:
**Describe the bug**
Unable to build sensor sample **bq274xx** within last origin/master checkout. Obviously it has something to do with CMake build:

`[62/143] Building C object zephyr/drivers/sensor/bq274xx/CMakeFiles/drivers__sensor__bq274xx.dir/bq274xx.c.obj 
FAILED: zephyr/drivers/sensor/bq274xx/CMakeFiles/drivers__sensor__bq274xx.dir/bq274xx.c.obj`

DTS files for recommended nrf9160_innblue21 board does not include any of bq27421 i2c register adresses so I tried to add that node but behavior haven't change.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b nrf9160_innblue21 samples/sensors/bq274xx

**Expected behavior**
Should build w/o errors.

**Logs and console output**
[Screen shot](https://imgur.com/TnH2erp)

**Environment ():**
 - OS:  Windows
 - Toolchain: gcc-arm-none-eabi-8-2018-q3-update
-- SHA1: 61d42cf42b3003e04c9e3c1c40aa612b8f18aa28

**Additional context**
Add any other context about the problem here.

MaureenHelm, 2020-09-23:
@parthitce will you please have a look?

parthitce, 2020-09-23:
@mejtejt It builds fine with `west build -b nrf9160_innblue22 -s ./samples/sensor/bq274xx/`.

`bq27421` is present only on `nrf9160_innblue22`, but not on `nrf9160_innblue21` if I am not wrong. If you are connecting it externally to `nrf9160_innblue21`, include 

```
        bq27421@55 {
                compatible = "ti,bq274xx";
                label = "BQ274XX";
                reg = <0x55>;
                design-voltage = <3700>;
                design-capacity = <1800>;
                taper-current = <45>;
                terminate-voltage = <3000>;
        };
```

in respective i2c node and also add `nrf9160_innblue21` to [platform_allow](https://github.com/zephyrproject-rtos/zephyr/blob/master/samples/sensor/bq274xx/sample.yaml#L7).

@MaureenHelm Not a bug, but wrong board.

mejtejt, 2020-09-23:
> @mejtejt It builds fine with `west build -b nrf9160_innblue22 -s ./samples/sensor/bq274xx/`.
>
> `bq27421` is present only on `nrf9160_innblue22`, but not on `nrf9160_innblue21` if I am not wrong. If [...]

That's right! Though, at README.rst says under **Requirements** : "- nrf9160_innblue21 board with BQ274XX sensor `BQ274XX Sensor`_"
Should be changed to nrf9160_innblue22 for clumsy reader, or explicitly stated externally connected.
Thak you both for your time

closed 2020-09-23 by mejtejt

MaureenHelm, 2020-09-23:
Thanks for the quick response!

PR #28641 (2020-10-21): sensor: bq27421: fix available platform in README
bq27421 is available only on nrf9160_innblue22

Fixes: #28638