zephyrproject-rtos/zephyr #28650: GCC-10.2 link issue w/g++ on aarch64

galak, 2020-09-23:
When trying to build `tests/application_development/libcxx` on aarch64 w/gcc-10.2 we get:

```
/home/galak/pr-270/aarch64-zephyr-elf/bin/../lib/gcc/aarch64-zephyr-elf/10.2.0/../../../../aarch64-zephyr-elf/bin/ld: /home/galak/pr-270/aarch64-zephyr-elf/bin/../lib/gcc/aarch64-zephyr-elf/10.2.0/libgcc.a(unwind-dw2.o): in function `uw_update_context':
unwind-dw2.c:(.text+0xe60): undefined reference to `strchr'
unwind-dw2.c:(.text+0xe60): relocation truncated to fit: R_AARCH64_CALL26 against undefined symbol `strchr'
```

This seems to be because of dependency between libgcc and libc.

galak, 2020-09-23:
@tejlmand would appreciate your help on this one.

On the link we have:
` -lstdc++ -lm -lc -lgcc`

If I tweak it by hand to be either:

` -lstdc++ -lm -Wl,--start-group -lc -lgcc -Wl,--end-group `

or

` -lstdc++ -lm -Wl,--start-group -lc -lgcc -lc`

it will work correctly.

pabigot, 2020-11-24:
Removed stale.  This is still present in 0.12.0-beta-2 @galak:
```
tirzah[433]$ rm -rf build && west build -b qemu_cortex_a53 
-- west build: generating a build system
Including boilerplate (Zephyr base): /mnt/nordic/zp/zephyr/cmake/app/boilerplate.cmake
-- Application: /mnt/nordic/zp/zephyr/tests/application_development/libcxx
-- Zephyr version: 2.4.99 (/mnt/nordic/zp/zephyr)
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.5") found components: Interpreter 
-- Found west (found suitable version "0.8.0", minimum required is "0.7.1")
-- Board: qemu_cortex_a53
-- Cache files will be written to: /home/pab/.cache/zephyr
-- Found toolchain: zephyr (/usr/local/zephyr-sdk-0.12.0-beta-2)
-- Found dtc: /usr/local/zephyr-sdk-0.12.0-beta-2/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /mnt/nordic/zp/zephyr/boards/arm/qemu_cortex_a53/qemu_cortex_a53.dts
qemu_cortex_a53.dts.pre.tmp:56.21-67.5: Warning (simple_bus_reg): /soc/timer: missing or empty reg/ranges property
-- Generated zephyr.dts: /mnt/nordic/zp/zephyr/tests/application_development/libcxx/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /mnt/nordic/zp/zephyr/tests/application_development/libcxx/build/zephyr/include/generated/devicetree_unfixed.h
Parsing /mnt/nordic/zp/zephyr/Kconfig
Loaded configuration '/mnt/nordic/zp/zephyr/boards/arm/qemu_cortex_a53/qemu_cortex_a53_defconfig'
Merged configuration '/mnt/nordic/zp/zephyr/tests/application_development/libcxx/prj.conf'
Configuration saved to '/mnt/nordic/zp/zephyr/tests/application_development/libcxx/build/zephyr/.config'
Kconfig header saved to '/mnt/nordic/zp/zephyr/tests/application_development/libcxx/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/local/zephyr-sdk-0.12.0-beta-2/aarch64-zephyr-elf/bin/aarch64-zephyr-elf-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /mnt/nordic/zp/zephyr/tests/application_development/libcxx/build
-- west build: building application
[1/101] Preparing syscall dependency handling

[96/101] Linking CXX executable zephyr/zephyr_prebuilt.elf
FAILED: zephyr/zephyr_prebuilt.elf 
: && /usr/local/zephyr-sdk-0.12.0-beta-2/aarch64-zephyr-elf/bin/aarch64-zephyr-elf-g++     zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj  -o zephyr/zephyr_prebuilt.elf  -Wl,-T  zephyr/linker.cmd  -Wl,-Map=/mnt/nordic/zp/zephyr/tests/application_development/libcxx/build/zephyr/zephyr_prebuilt.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/aarch64/libarch__arm__core__aarch64.a  zephyr/arch/arch/arm/core/aarch64/mmu/libarch__arm__core__aarch64__mmu.a  zephyr/lib/libc/newlib/liblib__libc__newlib.a  zephyr/lib/posix/liblib__posix.a  zephyr/subsys/testsuite/ztest/libsubsys__testsuite__ztest.a  zephyr/drivers/serial/libdrivers__serial.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj  -L"/usr/local/zephyr-sdk-0.12.0-beta-2/aarch64-zephyr-elf/bin/../lib/gcc/aarch64-zephyr-elf/10.2.0"  -L/mnt/nordic/zp/zephyr/tests/application_development/libcxx/build/zephyr  -lgcc  -Wl,--print-memory-usage  zephyr/arch/common/libisr_tables.a  -mcpu=cortex-a53  -mabi=lp64  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -no-pie  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -lstdc++  -lm  -lc  -lgcc  -L"/usr/local/zephyr-sdk-0.12.0-beta-2/aarch64-zephyr-elf/aarch64-zephyr-elf"/lib/. && :
/usr/local/zephyr-sdk-0.12.0-beta-2/aarch64-zephyr-elf/bin/../lib/gcc/aarch64-zephyr-elf/10.2.0/../../../../aarch64-zephyr-elf/bin/ld: /usr/local/zephyr-sdk-0.12.0-beta-2/aarch64-zephyr-elf/bin/../lib/gcc/aarch64-zephyr-elf/10.2.0/libgcc.a(unwind-dw2.o): in function `uw_update_context':
unwind-dw2.c:(.text+0xe60): undefined reference to `strchr'
unwind-dw2.c:(.text+0xe60): relocation truncated to fit: R_AARCH64_CALL26 against undefined symbol `strchr'
Memory region         Used Size  Region Size  %age Used
           FLASH:          0 GB        64 MB      0.00%
            SRAM:      218184 B    131056 KB      0.16%
        IDT_LIST:          56 B         2 KB      2.73%
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /mnt/nordic/zp/zephyr/tests/application_development/libcxx/build
```

galak, 2020-11-24:
@tejlmand ping, how should we fix this?

PR #30521 (2021-01-20): cmake: newlib linking flags -lc and -lgcc circular dependency
Fixes: #28650

Linking with newlib now defines the following linker flags as:
```
${CMAKE_C_LINKER_WRAPPER_FLAG}${CMAKE_LINK_LIBRARY_FLAG}c
${CMAKE_C_LINKER_WRAPPER_FLAG}${CMAKE_LINK_LIBRARY_FLAG}gcc
c
```

This is needed because when linking with newlib on aarch64, then libgcc
has a link dependency to libc (strchr), but libc also has dependencies
to libgcc.

CMake is capable of handling circular link dependencies for CMake
defined static libraries, which can be further controlled using
`LINK_INTERFACE_MULTIPLICITY`.

However, libc  [... cut]
commit da372600 (2021-01-20): cmake: newlib linking flags -lc and -lgcc circular dependency

Fixes: #28650

Linking with newlib now defines the following linker flags as:
```
${CMAKE_C_LINKER_WRAPPER_FLAG}${CMAKE_LINK_LIBRARY_FLAG}c
${CMAKE_C_LINKER_WRAPPER_FLAG}${CMAKE_LINK_LIBRARY_FLAG}gcc
c
```

This is needed because when linking with newlib on aarch64, then libgcc
has a link dependency to libc (strchr), but libc also has dependencies
to libgcc.

CMake is capable of handling circular link dependencies for CMake
defined static libraries, which can be further controlled using
`LINK_INTERFACE_MULTIPLICITY`.

However, libc [... cut]
closed 2021-01-20 by nashif