zephyrproject-rtos/zephyr #28653: Bluetooth: Mesh: TX Power Dynamic Control

MarcianoPreciado, 2020-09-23:
**Describe the bug**
The dynamic TX power control API enabled by `CONFIG_BT_CTLR_TX_PWR_DYNAMIC_CONTROL` is currently incompatible with Mesh.

**To Reproduce**
1. Copy the sample project `samples/bluetooth/mesh`
2. Add to prj.conf: 
```
CONFIG_BT_CTLR_ADVANCED_FEATURES=y
CONFIG_BT_CTLR_TX_PWR_DYNAMIC_CONTROL=y
```
3. Add these functions to main.c:
```
static void set_tx_power_adv(int8_t tx_pwr_lvl)
{
        uint8_t handle_type = BT_HCI_VS_LL_HANDLE_TYPE_ADV;
        uint16_t handle = NULL;
	struct bt_hci_cp_vs_write_tx_power_level *cp;
	struct bt_hci_rp_vs_write_tx_power_level *rp;
	struct net_buf *buf, *rsp = NULL;
	int err;

	buf = bt_hci_cmd_create(BT_HCI_OP_VS_WRITE_TX_POWER_LEVEL,
				sizeof(*cp));
	if (!buf) {
		printk("Unable to allocate command buffer\n");
		return;
	}

	cp = net_buf_add(buf, sizeof(*cp));
	cp->handle = sys_cpu_to_le16(handle);
	cp->handle_type = handle_type;
	cp->tx_power_level = tx_pwr_lvl;

	err = bt_hci_cmd_send_sync(BT_HCI_OP_VS_WRITE_TX_POWER_LEVEL,
				   buf, &rsp);
	if (err) {
		uint8_t reason = rsp ?
			((struct bt_hci_rp_vs_write_tx_power_level *)
			  rsp->data)->status : 0;
		printk("Set Tx power err: %d reason 0x%02x\n", err, reason);
		return;
	}

	rp = (void *)rsp->data;
	printk("Actual Tx Power: %d\n", rp->selected_tx_power);

	net_buf_unref(rsp);
}

static void set_tx_power_scan(int8_t tx_pwr_lvl)
{
        uint8_t handle_type = BT_HCI_VS_LL_HANDLE_TYPE_SCAN;
        uint16_t handle = NULL;
	struct bt_hci_cp_vs_write_tx_power_level *cp;
	struct bt_hci_rp_vs_write_tx_power_level *rp;
	struct net_buf *buf, *rsp = NULL;
	int err;

	buf = bt_hci_cmd_create(BT_HCI_OP_VS_WRITE_TX_POWER_LEVEL,
				sizeof(*cp));
	if (!buf) {
		printk("Unable to allocate command buffer\n");
		return;
	}

	cp = net_buf_add(buf, sizeof(*cp));
	cp->handle = sys_cpu_to_le16(handle);
	cp->handle_type = handle_type;
	cp->tx_power_level = tx_pwr_lvl;

	err = bt_hci_cmd_send_sync(BT_HCI_OP_VS_WRITE_TX_POWER_LEVEL,
				   buf, &rsp);
	if (err) {
		uint8_t reason = rsp ?
			((struct bt_hci_rp_vs_write_tx_power_level *)
			  rsp->data)->status : 0;
		printk("Set Tx power err: %d reason 0x%02x\n", err, reason);
		return;
	}

	rp = (void *)rsp->data;
	printk("Actual Tx Power: %d\n", rp->selected_tx_power);

	net_buf_unref(rsp);
}
```
4. Add the functions into `main()`
```
void main(void)
{
	int err;

	printk("Initializing...\n");

	/* Initialize the Bluetooth Subsystem */
	err = bt_enable(bt_ready);
        
       // Add tx gain settings
        static const uint8_t txp[] = {8, 4, 0, -4};
        uint8_t tx = txp[3];
        set_tx_power_adv(tx);
        set_tx_power_scan(tx);
	
        if (err) {
		printk("Bluetooth init failed (err %d)\n", err);
	}
}
```
**Expected behavior**
One would expect that these settings remain for the duration of the program, but rather are overwritten with the default values in what seems like less than 1 second.

**Impact**
This is quite a hinderance for my application which is a device which supports multiple roles depending on whether or not it is wired. I would like my relays to have 8dBm output power and my LPNs to have 0dBm and be able to choose the TX gain at runtime. I can make it work but it is a sloppy workaround. (More on that later)

**Environment (please complete the following information):**
 - OSX
 - Toolchain: gnuarmemb
 - commit 3a311235b2c4f60a42fbc5fb89a59166700bbf86

MarcianoPreciado, 2020-09-23:
**The Root Cause**
Because of the way mesh works, the function `ll_adv_enable` from `subsys/bluetooth/controller/ll_sw/ull_adv.c` gets called repeadedly and rapidly.
This function has the following lines (which appear twice! Ln 450, Ln 604) , which reset the TX gain back to the default.
```
#if defined(CONFIG_BT_CTLR_TX_PWR_DYNAMIC_CONTROL)
 	lll->tx_pwr_lvl = RADIO_TXP_DEFAULT;
 #endif /* CONFIG_BT_CTLR_TX_PWR_DYNAMIC_CONTROL */
```

MarcianoPreciado, 2020-09-23:
`subsys/bluetooth/controller/ll_sw/ull_scan.c`, `subsys/bluetooth/controller/ll_sw/ull_master.c`, and `subsys/bluetooth/controller/ll_sw/ull_slave.c` all have the same snippets in their equivalent initializer functions. I'm guessing only `ull_adv.c` and `ull_scan.c` effect the mesh functionality however.

MarcianoPreciado, 2020-09-23:
Looping in @rstoica and @carlescufi since they are the author and committer (respectively) of the dynamic tx power feature. Perhaps you guys could provide some insight?

MarcianoPreciado, 2020-09-23:
Perhaps there could be a static variable 
```
// Initialize as default
static uint8_t tx_power = RADIO_TXP_DEFAULT;
```
Make it so that it can only be set if `CONFIG_BT_CTLR_TX_PWR_DYNAMIC_CONTROL` is defined.

Then
```
#if defined(CONFIG_BT_CTLR_TX_PWR_DYNAMIC_CONTROL)
 	lll->tx_pwr_lvl = RADIO_TXP_DEFAULT;
 #endif /* CONFIG_BT_CTLR_TX_PWR_DYNAMIC_CONTROL */
```
can become 
```
lll->tx_pwr_lvl = tx_power;
```

cvinayak, 2020-09-24:
@MarcianoPreciado Do you have this: https://github.com/zephyrproject-rtos/zephyr/commit/1723e479ab6c1dfb6a18db0d908e31903fb34ba0

MarcianoPreciado, 2020-09-24:
@cvinayak I have been running on v2.3.0 with the nRF Connect SDK which is slightly behind. The above mentioned commit solves my problem. Thank you very much!

closed 2020-09-24 by MarcianoPreciado