zephyrproject-rtos/zephyr #28670: drivers: flash: bluetooth: stm32wb: attempt to erase internal flash before enabling Bluetooth cause fatal error

Zzzwierzak, 2020-09-24:
**Describe the bug**
When CONFIG_BT option is set, attempt to erase internal flash before enabling Bluetooth will cause fatal error.
Shci send driver API is initialized in open Bluetooth step, if flash erase command is executed before this step API is not yet set which cause error. 

**To Reproduce**
Steps to reproduce the behavior:
1. Open dir samples/drivers/flash_shell
2. Add CONFIG_BT=y, CONFIG_BT_CENTRAL=y to proj.conf
3. west build -p auto -b nucleo_wb55rg .
4. west flash
5. Open uart console
6. Send: flash erase 0xc8000 0x3000

**Expected behavior**
Flash partition erased. 

**Impact**
annoyance

**Logs and console output**
```
*** Booting Zephyr OS build v2.4.0-rc2-53-g61d42cf42b30  ***
Found flash controller FLASH_CTRL.
Flash I/O commands can be run.

 

uart:~$ flash erase 0xc8000 0x3000
[00:00:32.637,000] <err> os: ***** USAGE FAULT *****
[00:00:32.637,000] <err> os:   Illegal use of the EPSR
[00:00:32.637,000] <err> os: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0x20007821
[00:00:32.637,000] <err> os: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x08005713
[00:00:32.637,000] <err> os:  xpsr:  0x60000000
[00:00:32.637,000] <err> os: Faulting instruction address (r15/pc): 0x00000000
[00:00:32.637,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:32.637,000] <err> os: Current thread: 0x200006d0 (shell_uart)
[00:00:32.700,000] <err> os: Halting system

```

erwango, 2020-10-05:
@Zzzwierzak This behavior is the direct consequence of STM32WB Flash memory management as described in AN5289.
According to this, flash write/erase operations from CPU1 (where Zephyr is running) should be done after sending SHCI_C2_FLASH_EraseActivity(ERASE_ACTIVITY_ON) to CPU2. This cannot be done before SHCI init.

One way to workaround this issue (to be tested) would be to move CPU2 init out of `bt_ipm_open` and do it directly at driver init (in `_bt_ipm_init`). So SCHI init would then be done directly at start up.

Would it be of any use in your case ?

Zzzwierzak, 2020-10-06:
@erwango Can you provide example (or diff) what should be moved from open function to driver init? 

How big will be impact of this change on power consumption in low power states comparing to application that is not performing bt open(without proposed change)?

erwango, 2020-10-07:
@Zzzwierzak You can check https://github.com/erwango/zephyr/tree/fix_wb_early_c2_init for the code impact and have a test.
One consequence on power side is the activation of ABH3 bus clock. Though, I need to check more for the impacts on consumption in low power states.

erwango, 2020-10-07:
> I need to check more for the impacts on consumption in low power states.

One thing to consider on the LPM impact is the typical lifecycle of your product:
What we're doing with this change is to move C2 init from `bt_enable` to boot time. So power consumption will be the similar to what you get after `ble_enable`. Since, after bt_enable, there is no way to disable BLE stack (AFAIK), C2 will be enabled for the rest of the time the product is switched on. 
Does it make a difference in your product life cycle ? Do you happen to have period of time before  `bt_enable` significantly longer that after you run `bt_enable`?

This being said, I'm not finding any impacts as C2 should be taken care of in Low power modes.

Zzzwierzak, 2020-10-08:
@erwango, OK, I'm fine with this LPM behavior, I've tested proposed change and it works for me. 
One more question, when BT RF field is enabled?  Is this change cause activation of rf field during init stage?

erwango, 2020-10-09:
> Is this change cause activation of rf field during init stage?

It depends of what you understand by "rf field":
Yes, radio is now enabled at boot time with this change.
Though, RF activity will start only once `bt_enable` is called.

Zzzwierzak, 2020-10-09:
OK, will you merge proposed change to zephyr master branch?

erwango, 2020-10-09:
> OK, will you merge proposed change to zephyr master branch?

This is the plan, but I need to do some testing first.

closed 2020-11-19 by carlescufi