zephyrproject-rtos/zephyr #28685: Bluetooth: Characteristic unsubscribe under indication load results in ATT timeout

JordanYates, 2020-09-25:
**Describe the bug**

Conditions:
* Zephyr device acting as a GATT server
* Many indications are being generated in a loop in response to an event (accelerometer)
* GATT client attempts to unsubscribe from the indication characteristic while indications are queued

This sequence reliably results in an ATT timeout, normally due to a failure to allocate a buffer for the "write response" to the CCC write. The unsubscribe request in the below log is sent at ~00:02:07.
```
00> [00:02:03.931,365] <wrn> uc_bt_gatt: DATA: Subscribed 2
00> [00:02:32.000,091] <inf> act_ct_ctlr: 43 seconds to periodic packet
00> [00:02:37.906,616] <wrn> bt_conn: Unable to allocate buffer within timeout
00> [00:02:37.906,646] <err> bt_att: Unable to allocate buffer for op 0x13, 0
00> [00:02:37.919,097] <err> bt_att: ATT Timeout
00> [00:02:37.919,128] <wrn> uc_bt_gatt: DATA: Indication failed 14
00> [00:02:37.919,158] <wrn> uc_bt_gatt: DATA: Indication failed 14
00> [00:02:37.919,158] <wrn> uc_bt_gatt: DATA: Indication failed 14
00> [00:02:37.919,219] <wrn> uc_bt_gatt: DATA: Subscribed 0
00> [00:02:37.919,250] <err> bt_conn: not connected!
00> [00:02:37.919,281] <wrn> bt_att: No pending ATT request
00> [00:02:37.919,372] <err> bt_gatt: Error sending ATT PDU: -128
```

On the server, indications are broadcast normally to anyone that has subscribed through:
```
for (int i = 0; i < 10; i++) {
    indicate_params.attr = &uc_svc.attrs[DATA_ATTR_INDEX];
    indicate_params.data = message->data;
    indicate_params.len = message->len;
    indicate_params.func = uc_bt_data_indicate_cb;
    bt_gatt_indicate(NULL, &indicate_params);
}
```
With the `uc_bt_data_indicate_cb` printing the errors from the above log
```
if (err) {
	LOG_WRN("DATA: Indication failed %d", err);
}
```

Presumably all available TX buffers have been claimed by the indication loop, leaving no buffers for the write response.

Probably relevant Kconfig symbols
```
#define CONFIG_BT_CTLR_RX_BUFFERS 6
#define CONFIG_BT_CTLR_TX_BUFFERS 3
#define CONFIG_BT_CTLR_TX_BUFFER_SIZE 251
#define CONFIG_BT_CTLR_TX_PWR_PLUS_8 1
#define CONFIG_BT_CTLR_LE_ENC 1
#define CONFIG_BT_CTLR_CONN_PARAM_REQ 1
#define CONFIG_BT_CTLR_EXT_REJ_IND 1
#define CONFIG_BT_CTLR_SLAVE_FEAT_REQ 1
#define CONFIG_BT_CTLR_LE_PING 1
#define CONFIG_BT_CTLR_PRIVACY 1
#define CONFIG_BT_CTLR_RL_SIZE 8
#define CONFIG_BT_CTLR_EXT_SCAN_FP 1
#define CONFIG_BT_CTLR_DATA_LENGTH 1
#define CONFIG_BT_CTLR_DATA_LENGTH_MAX 251
#define CONFIG_BT_CTLR_MIN_USED_CHAN 1
#define CONFIG_BT_CTLR_CHAN_SEL_2 1
#define CONFIG_BT_HCI_CMD_COUNT 2
#define CONFIG_BT_RX_BUF_COUNT 3
#define CONFIG_BT_RX_BUF_LEN 76
#define CONFIG_BT_CONN_TX_MAX 3
#define CONFIG_BT_AUTO_DATA_LEN_UPDATE 1
#define CONFIG_BT_L2CAP_RX_MTU 252
#define CONFIG_BT_L2CAP_TX_BUF_COUNT 3
#define CONFIG_BT_L2CAP_TX_FRAG_COUNT 2
#define CONFIG_BT_L2CAP_TX_MTU 252
#define CONFIG_BT_ATT_ENFORCE_FLOW 1
#define CONFIG_BT_ATT_PREPARE_COUNT 0
#define CONFIG_BT_ATT_TX_MAX 3
```

**Expected behavior**
Server should respond with an acknowledgement to the CCC write, and any pending indications should be dropped.

While this is caused by insufficient `CONFIG_BT_CTLR_TX_BUFFERS`, it should be possible for the controller to iterate over queued packets on a characteristic if it is being unsubscribed from and dequeue them before trying to allocate the write response packet.

**Impact**
Annoyance, normally we don't unsubscribe until we are done with a connection.
While the connection is unusable after an ATT Timeout, reconnecting fixes the issue.

Increasing `CONFIG_BT_CTLR_TX_BUFFERS` from the default value also reduces the chances of the TX buffer starvation.

**Logs and console output**

Wireshark Capture of the server failing to respond (unacknowledged write is packet 1127)
[IndicateUnsub.zip](https://github.com/zephyrproject-rtos/zephyr/files/5280073/IndicateUnsub.zip)

**Environment (please complete the following information):**
 - Zephyr 2.4-rc3

carlescufi, 2020-10-01:
@JordanYates 

>While this is caused by insufficient CONFIG_BT_CTLR_TX_BUFFERS, it should be possible for the controller to iterate over queued packets on a characteristic if it is being unsubscribed from and dequeue them before trying to allocate the write response packet.

If the indications have been already sent to the controller from the host, then there is no way according to spec to flush them, and they *will* go out regardless. However, after the host sends the write response for the unsubscribe, the stack should not allow for any additional indications.

carlescufi, 2020-10-01:
@Vudentz any input on this one?

Vudentz, 2020-10-02:
> @Vudentz any input on this one?

Indication do require a response so the are not queued in the controller, instead we queue them the host and wait the other requests to complete.

There might be some problem with BT_ATT_TX_MAX though. if we can queue all buffers then we are likely to create such problems since one can exaust the buffer pool on TX alone the response also fail waiting for a free buffer, we could attempt to dequeue and free the buffer but I wonder if it wouldn't be a better idea to not use the same pool for requests and response, that way we can garantee responses do not have to wait for a buffer that is allocate for a request which has a much longer lifetime when queued.

JordanYates, 2020-12-09:
I have pushed a sample application that demonstrates this issue based on master.
https://github.com/csiro-wsn/zephyr/tree/bug/indication_timeout/samples/bluetooth/indication_timeout

The issue is reproducible using the nRF Connect Android application as the central.

PR #31184 (2021-02-11): Bluetooth: host: Fix ATT indication deadlock in RX thread
Fix deadlock in RX thread when ATT all buffers are taken by ATT requests and the RX thread blocks waiting for an buffer.
Solve this by not having a reference to the ATT request buffer until the ATT response is received, instead get a new ATT buffer in the case where it has to be resent.

Fixes: #28685

Fix ATT transaction violation in indications if the application has not set a callback function. In this case the indication was not treated as a transaction, leading to transaction violation.

Fix security elevation not working properly  [... cut]
closed 2021-02-11 by nashif