zephyrproject-rtos/zephyr #28694: k_wakeup follwed by k_thread_resume call causes system freeze

rhardik, 2020-09-25:
**Describe the bug**
I've added a while loop with 8 seconds sleep in a thread. Now by mistake I called both  k_wakeup and  k_thread_resume which is causing system to get freeze without any crash logs.

**To Reproduce**
Steps to reproduce the behaviour:
1. $ git clone -b thread-handling-issue https://github.com/rhardik/reproduce-zephyr-issues.git
2.$ cd _reproduce-zephyr-issues_
3. copy _issue-1_ folder to zephyr compilation environment
4. $ cd issue-1 && west build -b <board_name>    (I'm using nRF52 boards)
5. west flash
6. press button-0 to suspend thread. 
7. press button-1 to resume thread (here I've used both  k_wakeup and k_thread_resume)
8. See at UART terminal: system gets freeze
9. Now edit the code. Enable _k_thread_suspend(dummy_tid)_

` while (1) {
                        k_sleep(K_MSEC(8000));
                        printk("flag = %d\n",dummy_flg);
                        printk("%s\n", __func__);
                        //k_thread_suspend(dummy_tid);
        }
`
10. Recompile and flash the code.
11. Press button-0 to see if suspend thread works or not.
12. Press button-1 to resume the thread.
13. See at UART terminal:  while loop continuously running without considering _k_sleep(8000)_

**Expected behavior**
System should not get freeze if I call k_thread_resume after k_wakeup (Note: thread in sleep state here)

**Impact**
annoyance.

**Logs and console output**
[image]

**Environment (please complete the following information):**
 - OS: Linux)
 - Toolchain - Zephyr SDK
- Commit a3fae2f  (zephyr v2.2.99)

andyross, 2020-09-25:
We don't generally do defensive state checking in kernel APIs.  Indeed, resuming an unsuspended thread or waking up a non-sleeping one is a fatal error, and exactly the behavior you get is essentially undefined.

It's no different than trying to do things like double-remove a list item, etc...

I suppose someone could add some assertions in there to check state.

pabigot, 2020-09-25:
I confirm that `k_wakeup` and `k_thread_resume` do not play well together, and do not seem to be no-ops when the thing they're supposed to do is unnecessary.

https://github.com/pabigot/rhardik-reproduce-zephyr-issues/commits/issue/28694 has two commits, the first updating the example to current Zephyr APIs, and the second making the test a little nicer to work with.  Below is the summary of the functional changes and observed behavior.

Shorten the delay and print a running timestamp in the dummy thread.

Simplify compile-time selection of whether the dummy thread
self-suspends.

Use a counter to iterate through four responses to SW1 press:
* only k_wakeup
* only k_thread_resume
* both k_wakeup and k_thread_resume
* neither k_wakeup nor k_thread_resume

If the dummy thread does not self-suspend then k_wakeup alone, and k_thread_resume alone, both work.  When both are invoked, the program hangs.  This appears to be a bug.

If the dummy thread does self-suspend then
* SW1 #1 k_wakeup has no effect
* SW1 #2 k_thread_resume has no effect
* SW1 #3 k_wakeup and k_thread_resume releases the dummy thread and it  fast-loops, neither sleeping nor suspending.  This appears to be a bug.

andrewboie, 2020-10-15:
These are syscalls so even if we are lax about checking correct usage from supervisor mode contexts, a nefarious user thread shouldn't be able to cause corruption or DoS via the syscalls, at minimum needs to be 100% robust from user threads (and very likely worth making robust from all contexts). I'd say Medium priority is appropriate.

nashif, 2020-10-17:
>  Indeed, resuming an unsuspended thread or waking up a non-sleeping one is a fatal error, and exactly the behavior you get is essentially undefined.

that is not true per the documentation:

```
/**
 * @brief Wake up a sleeping thread.
 *
 * This routine prematurely wakes up @a thread from sleeping.
 *
 * If @a thread is not currently sleeping, the routine has no effect.
 *
 * @param thread ID of thread to wake.
 *
 * @return N/A
 */
__syscall void k_wakeup(k_tid_t thread);
```

and

```
/**
 * @brief Resume a suspended thread.
 *
 * This routine allows the kernel scheduler to make @a thread the current
 * thread, when it is next eligible for that role.
 *
 * If @a thread is not currently suspended, the routine has no effect.
 *
 * @param thread ID of thread to resume.
 *
 * @return N/A
 */
__syscall void k_thread_resume(k_tid_t thread);
```

Both calls should just have no effect, it should not hang or crash.

PR #29299 (2020-10-22): Only resume suspended threads
- only resume suspended threads
- Do not queue a thread that is already queued

Fixes #28694
Fixes #29244
commit bf69afcd (2020-10-22): kernel: only resume suspended threads

Do not try to resume a thread that was not suspended.

Fixes #28694
closed 2020-10-22 by nashif