zephyrproject-rtos/zephyr #28695: storage: FCB2 support

de-nordic, 2020-09-25:
**Is your feature request related to a problem? Please describe.**
The Mynewt has added new generation of FCB in 1.6.0 (https://cwiki.apache.org/confluence/display/MYNEWT/RN-1.6.0#RN1.6.0-FCB2-secondgenerationFCB) that has several improvements over FCB, quoting from release notes:
_**"Redesigned FCB with smaller RAM footprint, more failure resistant and better abstraction.'**_

**Describe the solution you'd like**
Import code from Mynewst source repo:
https://github.com/apache/mynewt-core/tree/master/fs/fcb2
Note that the FCB2 is not meta data compatible with FCB1, which means that the support should be parallel rather than replacing FCB1 with FCB2, at least for a few release to let users move on.
Additionally the FCB2 needs to be adjusted to use non-0xFF erase value flash devices.

**Describe alternatives you've considered**
Not doing it.

**Additional context**
Mynewt repo with fcb2: https://github.com/apache/mynewt-core/tree/master/fs/fcb2
Mynewt release notes: https://cwiki.apache.org/confluence/display/MYNEWT/RN-1.6.0#RN1.6.0-FCB2-secondgenerationFCB

Laczen, 2020-09-25:
@de-nordic, nvs was created to avoid one major failure possibility in fcb: when writing the length of a new entry goes wrong it is difficult to recover from this. I don't know if this was fixed in fcb2: it would require at startup to go to the end of the fs, check if it is ok, if not copy what is good to the next sector, erase the old sector, copy back what is good to the old sector and erase the temporarily used next sector.

de-nordic, 2020-09-25:
@Laczen thanks for info. This needs to be verified within FCB2 code.

abxy, 2022-01-27:
I am considering taking on porting FCB2.

From what I gather, FCB2 stores metadata in fixed length structs with a separate CRC in the top of each sector, allowing corrupt entries to be skipped easily. I think this should address your concerns @Laczen.

Laczen, 2022-01-27:
@abxy, I don't know how error proof fcb2 is compared with fcb/nvs. Nvs has some quite extensive testing to check all failure modes, probably fcb2 would need the same. I no longer consider nvs as a good solution as it requires non sequential flash writing. From your remark I would conclude that fcb2 also does not do sequential flash writing.

Laczen, 2022-01-27:
@abxy, I did a quick read of `fcb2_append`, it seems that fcb2 stores entry data (size, ...) from the end of a sector, while the data is from the start. It seems that the entry data is 6 bytes that might require byte write support in flash (the same as for fcb). As the data is written after the entry also the data needs to be protected with a 2 byte crc.
The major limitation IMO is that it requires byte write capability.

abxy, 2022-01-27:
> @abxy, I don't know how error proof fcb2 is compared with fcb/nvs. Nvs has some quite extensive testing to check all failure modes, probably fcb2 would need the same. I no longer consider nvs as a [...]

@Laczen I agree about testing. What is the issue with non-sequential flash writing?

> @abxy, I did a quick read of `fcb2_append`, it seems that fcb2 stores entry data (size, ...) from the end of a sector, while the data is from the start. It seems that the entry data is 6 bytes that [...]

I don't believe this is correct. It seems to accommodate other granularities via `fcb2_len_in_flash`  defined in [`fcb2/src/fcb_priv.h` ](https://github.com/apache/mynewt-core/blob/2673c6c414f05028abf12039033d3ec27f87d98b/fs/fcb2/src/fcb_priv.h)

Laczen, 2022-01-27:
> @Laczen I agree about testing. What is the issue with non-sequential flash writing?

More and more flashes (e.g. nand based) do not support non-sequential writing.
 
> I don't believe this is correct. It seems to accommodate other granularities via `fcb2_len_in_flash` defined in [`fcb2/src/fcb_priv.h` ](https://github.com/apache/mynewt-core/blob/2673c6c414f05028abf12039033d3ec27f87d98b/fs/fcb2/src/fcb_priv.h)

Correct.

nvlsianpu, 2022-02-22:
I missed objectives and requirements for consideration on introduction FCB2
One of possible objective might be a LOG backend
Requirements that we might consider might be the `sequential writing scheme` which makes the solution pretty fast, data-integrity check, ability to recovery the storage instance after a write failure, capability for invalidate records.

Looking through the conversation above it looks like porting of the FCB2 meets neither objective nor requirements.

de-nordic, 2024-02-23:
Closing due to lack of interest.

closed 2024-02-23 by de-nordic