zephyrproject-rtos/zephyr #28699: Bluetooth: controller: Speed up disconnect process when slave latency is used

sigurdnev, 2020-09-25:
When a BLE peripheral device calls bt_conn_disconnect() the time it takes until the disconnect packet(LL_TERMINATE_IND) is sent depends on if slave latency is used or not. If slave latency is used, the time until the packet is sent can take [connection-interval(ms) x #slave latency] extra time. It can therefore take several seconds before the disconnect packet is sent.

For devices that are being powered down, the disconnect process should not take several seconds. If the device loses power before the disconnect packet is sent, some peer devices would need to wait until the supervision timeout occurs until it can connect to another device. In addition to this, a long disconnect process also affects how fast a reconnection can happen. These are things that have a big impact on the user experience, especially for BLE HID devices.

Suggested fix: Don't wait for the slave latency time to elapse before sending the disconnect packet to the master device.

cvinayak, 2020-09-25:
This is a bug, as any non empty PDU enqueued shall break slave latency. Will analyse and provide a fix in the coming days.

cvinayak, 2020-09-28:
@sigurdnev Please try this #28725 and leave your comments in the PR.

PR #28725 (2020-10-01): Bluetooth: controller: Fix latency cancel on LLCP initiation
Fix controller implementation to cancel peripheral latency
when initiating new control procedures.

Fixes #28699.
commit 416d7ee3 (2020-10-01): Bluetooth: controller: Fix latency cancel on LLCP initiation

Fix controller implementation to cancel peripheral latency
when initiating new control procedures.

Fixes #28699.
closed 2020-10-01 by carlescufi