zephyrproject-rtos/zephyr #28706: west build -p auto -b nrf52840dk_nrf52840 error: HAS_SEGGER_RTT

mfinmuch, 2020-09-26:
I try to build the project under the zephyr/tests/bluetooth/ ,but it build failed
I tried to build helloword  ,it success
`neat@ubuntu:~/zephyrafh/zephyr/tests/bluetooth/afh_eval_latency$ west build -p auto -b nrf52840dk_nrf52840
-- west build: generating a build system
-- Application: /home/neat/zephyrafh/zephyr/tests/bluetooth/afh_eval_latency
-- Zephyr version: 2.4.0-rc3 (/home/neat/zephyrafh/zephyr)
-- Found west (found suitable version "0.7.3", minimum required is "0.7.1")
-- Board: nrf52840dk_nrf52840
-- Cache files will be written to: /home/neat/.cache/zephyr
-- Found toolchain: zephyr (/home/neat/zephyr-sdk)
-- Found dtc: /home/neat/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
CMake Warning at /home/neat/zephyrafh/zephyr/cmake/generic_toolchain.cmake:26 (message):
  ZEPHYR_GCC_VARIANT is deprecated, please use ZEPHYR_TOOLCHAIN_VARIANT
  instead
Call Stack (most recent call first):
  /home/neat/zephyrafh/zephyr/cmake/app/boilerplate.cmake:589 (include)
  CMakeLists.txt:4 (include)

-- Found BOARD.dts: /home/neat/zephyrafh/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts
-- Generated zephyr.dts: /home/neat/zephyrafh/zephyr/tests/bluetooth/afh_eval_latency/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/neat/zephyrafh/zephyr/tests/bluetooth/afh_eval_latency/build/zephyr/include/generated/devicetree_unfixed.h
Parsing /home/neat/zephyrafh/zephyr/Kconfig
Loaded configuration '/home/neat/zephyrafh/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840_defconfig'
Merged configuration '/home/neat/zephyrafh/zephyr/tests/bluetooth/afh_eval_latency/prj.conf'

error: HAS_SEGGER_RTT (defined at drivers/debug/Kconfig.rtt:4) is assigned in a configuration file,
but is not directly user-configurable (has no prompt). It gets its value indirectly from other
symbols. See http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_HAS_SEGGER_RTT.html
and/or look up HAS_SEGGER_RTT in the menuconfig/guiconfig interface. The Application Development
Primer, Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual
might be helpful too.

CMake Error at /home/neat/zephyrafh/zephyr/cmake/kconfig.cmake:239 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /home/neat/zephyrafh/zephyr/cmake/app/boilerplate.cmake:591 (include)
  CMakeLists.txt:4 (include)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/home/neat/zephyrafh/zephyr/tests/bluetooth/afh_eval_latency/build -S/home/neat/zephyrafh/zephyr/tests/bluetooth/afh_eval_latency -GNinja
` 
also I try to setting west build -t menuconfig,but it also failed

`neat@ubuntu:~/zephyrafh/zephyr/tests/bluetooth/afh_eval_latency$ west build -t menuconfig
-- west build: generating a build system
-- Application: /home/neat/zephyrafh/zephyr/tests/bluetooth/afh_eval_latency
-- Zephyr version: 2.4.0-rc3 (/home/neat/zephyrafh/zephyr)
-- Found west (found suitable version "0.7.3", minimum required is "0.7.1")
-- Board: nrf52840dk_nrf52840
-- Cache files will be written to: /home/neat/.cache/zephyr
-- Found toolchain: zephyr (/home/neat/zephyr-sdk)
-- Found dtc: /home/neat/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
CMake Warning at /home/neat/zephyrafh/zephyr/cmake/generic_toolchain.cmake:26 (message):
  ZEPHYR_GCC_VARIANT is deprecated, please use ZEPHYR_TOOLCHAIN_VARIANT
  instead
Call Stack (most recent call first):
  /home/neat/zephyrafh/zephyr/cmake/app/boilerplate.cmake:589 (include)
  CMakeLists.txt:4 (include)

-- Found BOARD.dts: /home/neat/zephyrafh/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts
-- Generated zephyr.dts: /home/neat/zephyrafh/zephyr/tests/bluetooth/afh_eval_latency/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/neat/zephyrafh/zephyr/tests/bluetooth/afh_eval_latency/build/zephyr/include/generated/devicetree_unfixed.h
Parsing /home/neat/zephyrafh/zephyr/Kconfig
Loaded configuration '/home/neat/zephyrafh/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840_defconfig'
Merged configuration '/home/neat/zephyrafh/zephyr/tests/bluetooth/afh_eval_latency/prj.conf'

error: HAS_SEGGER_RTT (defined at drivers/debug/Kconfig.rtt:4) is assigned in a configuration file,
but is not directly user-configurable (has no prompt). It gets its value indirectly from other
symbols. See http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_HAS_SEGGER_RTT.html
and/or look up HAS_SEGGER_RTT in the menuconfig/guiconfig interface. The Application Development
Primer, Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual
might be helpful too.

CMake Error at /home/neat/zephyrafh/zephyr/cmake/kconfig.cmake:239 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /home/neat/zephyrafh/zephyr/cmake/app/boilerplate.cmake:591 (include)
  CMakeLists.txt:4 (include)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/home/neat/zephyrafh/zephyr/tests/bluetooth/afh_eval_latency/build -S/home/neat/zephyrafh/zephyr/tests/bluetooth/afh_eval_latency -GNinja
`
Host and SDK deytails:
Ubuntu 18.04
Zephyr SDK sdk-0.11.3

mfinmuch, 2020-09-29:
I remove HAS_SEGGER_RTT in prj.conf then it build success 
but I don't know why

mbolivar-nordic, 2020-09-29:
Hi there! I see this in your build output:

> -- Application: /home/neat/zephyrafh/zephyr/tests/bluetooth/afh_eval_latency

Since this is not the name of any of the samples provided with zephyr, I'm guessing this is a custom app you wrote yourself.

> error: HAS_SEGGER_RTT (defined at drivers/debug/Kconfig.rtt:4) is assigned in a configuration file,
but is not directly user-configurable (has no prompt). It gets its value indirectly from other
symbols. See http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_HAS_SEGGER_RTT.html
and/or look up HAS_SEGGER_RTT in the menuconfig/guiconfig interface. The Application Development
Primer, Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual
might be helpful too.

This error is telling you what the problem is. You cannot set `CONFIG_HAS_SETTER_RTT=y` in a configuration file because it "is not directly user-configurable". The documentation the error message references explains in more detail.

There should be no need to set this option manually, so since this doesn't look like a bug in a prj.conf that's part of Zephyr, I'm going to close this issue. Please reopen if the app is in fact part of an application that is provided by the Zephyr project itself.

closed 2020-09-29 by mbolivar-nordic