zephyrproject-rtos/zephyr #28709: phandle-array does allow array of just phandles

ChaosinaCan, 2020-09-26:
**Describe the bug**

https://github.com/zephyrproject-rtos/zephyr/blob/0a0cb52fb2290b825f4e6b0d8d1f7bfad358cf01/dts/binding-template.yaml#L157-L160 indicates that in a value of type phandle-array, numbers are optional, so `<&foo &bar>` should be valid.

However, https://github.com/zephyrproject-rtos/zephyr/blob/0a0cb52fb2290b825f4e6b0d8d1f7bfad358cf01/scripts/dts/edtlib.py#L1225-L1230  appears to only allow `TYPE_PHANDLE` (a single phandle) and `TYPE_PHANDLES_AND_NUMS`. Is that supposed to be `TYPE_PHANDLES` instead of `TYPE_PHANDLE`?

**To Reproduce**
Steps to reproduce the behavior:

1. [Set up the ZMK keyboard firmware project](https://zmkfirmware.dev/docs/dev-setup).
2. Edit `app/boards/shields/romac/romac.keymap` and change one of the `bindings` arrays to contain only values such as `&reset` and `&trans` so there are no numbers in the array.
3. Build with 
```
cd app
west build -b proton_c -- -DSHIELD=romac
```

This will fail with an error like

> devicetree error: expected property 'bindings' in /keymap/nav_layer in proton_c.dts.pre.tmp to be assigned with 'bindings = < &foo 1 2 ... &bar 3 4 ... >' (a mix of phandles and numbers), not 'bindings = < &reset &trans &trans &trans &trans &trans  &trans &trans &trans  &trans &trans &trans >;'

**Expected behavior**
As far as I can tell, an array of just phandles should be valid for a phandle-array.

**Impact**
Annoyance. Can be worked around by adding something that uses numbers somewhere in the array.

**Environment (please complete the following information):**
 - OS: Windows 10
 - Toolchain: GNU Arm Embedded
 - Commit SHA or Version used: https://github.com/petejohanson/zephyr/tree/zmk-v2.3.0-with-fixes

henrikbrixandersen, 2020-09-27:
How about `<&foo>, <&bar>`?

ChaosinaCan, 2020-09-27:
Same error (tested in a different keymap with a slightly different array):

> devicetree error: expected property 'bindings' in /keymap/daw_layer in nice_nano.dts.pre.tmp to be assigned with 'bindings = < &foo 1 2 ... &bar 3 4 ... >' (a mix of phandles and numbers), not 'bindings = < &none >, < &none >, < &none >, < &none >, < &none >, < &none >, < &none >, < &none >, < &trans >, < &none >, < &none >, < &none >, < &none >, < &none >, < &none >, < &none >;'

b0661, 2020-09-28:
https://github.com/zephyrproject-rtos/zephyr/blob/0a0cb52fb2290b825f4e6b0d8d1f7bfad358cf01/dts/binding-template.yaml#L157-L160 the ´(possibly)´ relates to ´32-bit´ (and *not* to numbers). phandle-array is always a combination of phandle and associated number(s). The numbers may be up to 32-bit size. In case you only have phandles just use ´phandles´ in the binding.

´(possibly)´ should possibly replaced or removed.

ChaosinaCan, 2020-09-28:
> phandle-array is always a combination of phandle and associated number(s).

I'll admit I am still very unfamiliar with DT, but I'm confused as to why edtlib.py allows phandle-array to be either `TYPE_PHANDLE` or `TYPE_PHANDLES_AND_NUMS` then.

Having phandles with no associated numbers seems to work as long as there is at least one number somewhere in the array. If phandle-array allows any element to be a phandle with no associated number, why is it not valid for the entire array to be made of such elements? If that is not valid, then should phandle-array _require_ every phandle to have at least one associated number?

I don't know if DT was intended to be used like this, but one of the ways it is used in ZMK is to define a mapping between keyboard keys and the behaviors to use when each key is pressed. Some behaviors such as `&kp KEY` and `&lt LAYER KEY` have one or more associated numbers. Other behaviors such as `&reset` and `&none` don't have any associated numbers. ZMK allows you to mix both types of behaviors in a keymap.

b0661, 2020-09-29:
> Having phandles with no associated numbers seems to work as long as there is at least one number somewhere in the array. If phandle-array allows any element to be a phandle with no associated number, why is it not valid for the entire array to be made of such elements?

You are right, the error check seems too strict for the degenerated case that you describe. 
https://github.com/zephyrproject-rtos/zephyr/blob/adb62ea290ee4d127c5134630489f64c43c252ef/scripts/dts/edtlib.py#L1229

I could not find `behaviors.dtsi` which defines the behavior nodes. The reset and none nodes should have properties like `xxx-cells = 0;` that define the count of values to be 0. With this given the phandles list as a degenerated phandles-array is a valid construct.

In my view TYPE_PHANDLES should be added to the allowed dtslib types for `phandle-array`.

ChaosinaCan, 2020-09-30:
> I could not find behaviors.dtsi which defines the behavior nodes. The reset and none nodes should have properties like xxx-cells = 0; that define the count of values to be 0.

Yep. See https://github.com/zmkfirmware/zmk/blob/main/app/dts/behaviors/none.dtsi for example.

Should `TYPE_PHANDLE` be kept and `TYPE_PHANDLES` added, or should that be replaced with `TYPE_PHANDLES`?

b0661, 2020-10-01:
> Should TYPE_PHANDLE be kept and TYPE_PHANDLES added

Yes. Looking at https://github.com/zephyrproject-rtos/zephyr/blob/1342d12500867d3de09025037e65e756d5486eba/scripts/dts/dtlib.py#L1353-L1357 suggests that TYPE_PHANDLE and TYPE_PHANDLES are valid types.

mbolivar-nordic, 2020-10-02:
> TYPE_PHANDLE and TYPE_PHANDLES are valid types.

Agreed; I'll send a PR

PR #28927 (2020-10-06): scripts: dts: fix handling for phandle-arrays with *-cells=0
Each controller node in a phandle-array may set the number of cells in
a specifier as any nonnegative integer. Currently, we don't allow
this in edtlib in the case where there are multiple controllers in a
phandle-array property all of which have 0 cells in the relevant
specifier, which is not correct. Fix this, add a regression test, and
improve the error message while we are here.

Fixes: #28709
commit 1ea7bf0d (2020-10-06): scripts: dts: fix handling for phandle-arrays with *-cells=0

Each controller node in a phandle-array may set the number of cells in
a specifier as any nonnegative integer. Currently, we don't allow
this in edtlib in the case where there are multiple controllers in a
phandle-array property all of which have 0 cells in the relevant
specifier, which is not correct. Fix this, add a regression test, and
improve the error message while we are here.

Fixes: #28709
closed 2020-10-06 by galak