zephyrproject-rtos/zephyr #28714: Bluetooth: PTS upper tester: GAP/CONN/NCON/BV-02-C Fails because of usage of NRPA

kromancer, 2020-09-27:
**Describe the bug**
The implementation of GAP/CONN/NCON/BV-02-C in PTS v7.6.2
needs non-connectable undirected advertising,
with the IUT using its identity address.

Currently, our upper tester will use `bluetooth.h::BT_LE_ADV_NCONN`, which means NRPA, for all test cases that require non-connectable undirected advertising.

The test will fail as the lower tester will use a whitelist with the IUT's identity address.

**To Reproduce**
Run GAP/CONN/NCON/BV-02-C and observe failure.
An air trace reveals that the IUT is using an NRPA.
Check the test's HCI trace (attached) and observe that the lower is using a whitelist with the IUT's id address.

**Expected behavior**
The upper tester should not instruct the IUT to use an NRPA unless explicitly stated in a test's spec.
wid_54: `TSC_MMI_iut_send_advertising_report_event_0203_general_discoverable`
should not have the same effect as
wid_79: `TSC_MMI_iut_send_advertising_report_event_non_resolvable_address`

**Impact**
The conformance test fails when run with auto-pts.

**Logs and console output**
[GAP_CONN_NCON_BV_02_C_2020_09_25_16_53_51.zip](https://github.com/zephyrproject-rtos/zephyr/files/5288123/GAP_CONN_NCON_BV_02_C_2020_09_25_16_53_51.zip)

**Environment (please complete the following information):**
- PTS v7.6.2
- zephyr v2.4.0-rc3
- auto-pts @ 5c1f50f

joerchan, 2020-09-29:
Need to use BT_LE_ADV_OPT_USE_IDENTITY for non-connectable advertiser to use the identity address.

@kromancer Are you reporting a bug on the tester now?

kromancer, 2020-09-29:
@joerchan Yes, I am reporting a, low priority IMHO, bug on the tester.

The offending line, which makes the wrong assumption that all non connectable advertising should use an NRPA, is here:
https://github.com/zephyrproject-rtos/zephyr/blob/zephyr-v2.4.0/tests/bluetooth/tester/src/gap.c#L299

carlescufi, 2020-10-01:
@kromancer will you send a PR?

ryanjh, 2020-10-22:
GAP/CONN/NCON/BV-02-C passed after enabling CONFIG_BT_PRIVACY. This issue can be closed.

As this test is not using extended advertising, it won't change set a new advertising random address.
https://github.com/zephyrproject-rtos/zephyr/blob/zephyr-v2.4.0/subsys/bluetooth/host/hci_core.c#L778

joerchan, 2020-10-22:
Thanks @ryanjh

closed 2020-10-22 by joerchan