zephyrproject-rtos/zephyr #28722: Bluetooth: provide `struct bt_conn` to ccc_changed callback

JordanYates, 2020-09-28:
**Is your enhancement proposal related to a problem? Please describe.**

It is currently not possible from application code for a CCC write to trigger events on that same connection.
The current callback does not contain enough information for this to occur.
The current arguments are the attr that changed (the CCC descriptor itself) and its new value.

Each CCC descriptor is unique per connection, per the Bluetooth Core Specification (3.3.3.3):
```
In the case where multiple ATT bearers from the same device are supported by the GATT server,
each ATT bearer shall be considered to have a separate GATT client instance. Therefore each GATT
client shall have a separate Client Characteristic Configuration.
```

Therefore applications should be notified of **which** connection has written to the CCC descriptor.

**Describe the solution you'd like**

`struct bt_conn` should be provided to the `ccc_changed ` callback.

JordanYates, 2020-09-28:
Given that "just use `ccc_write`" is a valid answer to this request, I will await feedback before implementing anything.

JordanYates, 2020-10-24:
Per slack, manually constructing the CCC characteristic is the way forward.
```
BT_GATT_CCC_MANAGED(((struct _bt_gatt_ccc[])
					    { BT_GATT_CCC_INITIALIZER(NULL, uc_bt_ccc_data_written, NULL) }),
					   BT_GATT_PERM_READ | BT_GATT_PERM_WRITE)
```
`ccc_changed` is not useful for applications that care about the actual changed value of the CCC characteristic per connection, as all connections are grouped together.

closed 2020-10-24 by JordanYates

JordanYates, 2021-06-02:
One downside of using `ccc_write` instead of `ccc_changed`  that I realised later is that the GATT layer doesn't recognize that the characteristic has been subscribed at that point in the function.

i.e. you can't call `bt_gatt_notify` inside the `ccc_write` callback.

ChumKiu, 2025-07-14:
I've a peripheral device that have to get two connection from two central, one central have to activate some standard services and another one that have to activate a custom service, how to know what is the connection that have activated the custom service if into the bss_ccc_cfg_changed I do not know what service have selected the CCC inside my custom service?
Thanks for any hint.
BR