zephyrproject-rtos/zephyr #28723: Does not respect python virtualenv

benjizhai, 2020-09-28:
**Describe the bug**
When I run west build from virtualenv, it still detects system python instead of the virtualenv one.

**To Reproduce**
Steps to reproduce the behavior:
1. activate python venv
2. `west build -b auto -b nrf52840dongle_nrf52840 samples/basic/blinky`

**Logs and console output**
[1/125] Generating include/generated/kobj-types-enum.h, include/generated/otype-to-str.h FAILED: zephyr/include/generated/kobj-types-enum.h zephyr/include/generated/otype-to-str.h cd /home/benji/proj/zephyr/zephyr/build/zephyr && **/usr/bin/python3.8** /home/benji/proj/zephyr/zephyr/scripts/gen_kobject_list.py --kobj-types-output /home/benji/proj/zephyr/zephyr/build/zephyr/include/generated/kobj-types-enum.h --kobj-otype-output /home/benji/proj/zephyr/zephyr/build/zephyr/include/generated/otype-to-str.h --kobj-size-output /home/benji/proj/zephyr/zephyr/build/zephyr/include/generated/otype-to-size.h --include /home/benji/proj/zephyr/zephyr/build/zephyr/misc/generated/struct_tags.json Traceback (most recent call last): File "/home/benji/proj/zephyr/zephyr/scripts/gen_kobject_list.py", line 62, in <module> import elftools ModuleNotFoundError: No module named 'elftools'

**Environment (please complete the following information):**
 - Ubuntu 20.04

galak, 2020-09-29:
@tejlmand / @mbolivar-nordic thoughts?

mbolivar-nordic, 2020-09-29:
dupe of https://github.com/zephyrproject-rtos/west/issues/443

closed 2020-09-29 by mbolivar-nordic