zephyrproject-rtos/zephyr #28729: ARM:

ioannisg, 2020-09-28:
### ARM Architecture related issues tentatively scheduled for v2.5 release

**New features**
- [ ] [Cortex-M] QEMU support for ARM Cortex-M4F (#23185)

**Feature Enhancements**

- [ ] [Enh. ARM Cortex-M] Add HW stack protection during boot, too.
- [ ] [ARM Cortex-M, ARC, Kernel] Consider moving the arm_core_mpu API above the arch- level, so as not to duplicate for ARM, ARC
- [ ] [Enh. Kernel] Rename `userlocal_data `to something that does not contain "user" in the name, as it is used under `CONFIG_USERSPACE ` but beyond user threads
- [ ] [Enh. Cortex-R] Support floating-point (#TBD)
- [ ] [Enh. ARM] Cross-ARCH API for setting IRQ priority (RFC, #25910)

**Bug Fixes**
- [ ] [Bug ARM Cortex-M] Support for HW stack protection features together with CONFIG_ARM_SECURE_FIRMWARE=y and CONFIG_FP_SHARING (currently broken)

bbolen, 2020-10-07:
I'm not sure if you saw #28231 , but it may be worth considering for 2.5

ioannisg, 2020-10-07:
> I'm not sure if you saw #28231 , but it may be worth considering for 2.5

Sure, pls, just add this to the above list. (This ticket is only a meta issue for our convenience, to have the overview of what's done in each release)

ioannisg, 2021-02-08:
Moving all these to 2.6 release

galak, 2021-06-02:
@ioannisg can you clear this out / migrate to a 2.7 version.

ioannisg, 2021-06-02:
> @ioannisg can you clear this out / migrate to a 2.7 version.

Yeap, already started doing that.

ioannisg, 2021-06-11:
@galak moving remaining tasks in the ticket for 2.7 release.

closed 2021-06-11 by ioannisg