zephyrproject-rtos/zephyr #28773: Lower Link Layer code use upper link layer function have " undefined reference to"

mfinmuch, 2020-09-29:
I tried to do Bluetooth test on nrf52840
Under zephyr/subsys/bluetooth/controller/ll_sw, I have modified some programs and added modules "neat.c" and "neat.h" by myself
 I use "lll_conn.c" in zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll to reference the function(the function name is upd_conn_stat ) in "neat.c"
I have include "neat.h" in "lll_conn.c"
And after `west build -p auto -b nrf52840_pca10056`
the terminal show
`zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_conn.c:376: undefined reference to `upd_conn_stat'`

I think the reason is I use upper link layer(ull) function ("neat.h") in lower link Layer(lll)
How can i fix this problem?

cvinayak, 2020-09-29:
@mfinmuch Have you added your files to CMakeLists.txt file?

mfinmuch, 2020-09-29:
> @mfinmuch Have you added your files to CMakeLists.txt file?
Thanks!!!! I am a bit unfamiliar with cmakefile.txt
I add 
`zephyr_library_sources(
        ll_sw/neat.c
      )
      zephyr_library_sources_ifdef(
        CONFIG_BT_LLL_VENDOR_NORDIC
        ll_sw/nordic/lll/lll_master.c
      )
`
in 'if(CONFIG_BT_CENTRAL)'
and it finally build completed!!!!!!
There are lots of CMakeLists.txt in every folder but I  added is under /zephyr/subsys/bluetooth/controller

pabigot, 2020-09-30:
From what I can tell this has been resolved.  Please re-open if that's not true.

closed 2020-09-30 by pabigot