zephyrproject-rtos/zephyr #28785: shell: It should be possible to get list of commands without pressing tab

eugmes, 2020-09-29:
**Is your enhancement proposal related to a problem? Please describe.**
Entering TAB or any special characters is very awkward on some terminal emulators, for example those running on Android devices. Pressing TAB looks like the only way to get list of commands supported by shell.

**Describe the solution you'd like**
It should be possible to get list of supported command some other way. For example, by listing them in the output of the `help` command.

**Describe alternatives you've considered**
It may be possible to add some other command, but it will not be so easily discoverable and may conflict with commands implemented by other users. One can also use `?`.

jukkar, 2020-09-30:
You can always give `-h` to the sub-command like this:
```
uart:~$ net -h
net - Networking commands
Subcommands:
  allocs     :Print network memory allocations.
  arp        :Print information about IPv4 ARP cache.
  conn       :Print information about network connections.
  dns        :Show how DNS is configured.
  gptp       :Print information about gPTP support.
  iface      :Print information about network interfaces.
  ipv6       :Print information about IPv6 specific information and
              configuration.
  mem        :Print information about network memory usage.
  nbr        :Print neighbor information.
  ping       :Ping a network host.
  pkt        :net_pkt information.
  ppp        :PPP information.
  resume     :Resume a network interface
  route      :Show network route.
  stacks     :Show network stacks information.
  stats      :Show network statistics.
  suspend    :Suspend a network interface
  tcp        :Connect/send/close TCP connection.
  vlan       :Show VLAN information.
  websocket  :Print information about WebSocket connections.
uart:~$ net tcp -h
tcp - Connect/send/close TCP connection.
Subcommands:
  connect  :'net tcp connect <address> <port>' connects to TCP peer.
  send     :'net tcp send <data>' sends data to peer using TCP.
  recv     :'net tcp recv' receives data using TCP.
  close    :'net tcp close' closes TCP connection.
uart:~$ net tcp send -h
send - 'net tcp send <data>' sends data to peer using TCP.
uart:~$ 
```

eugmes, 2020-09-30:
@jukkar Yes, that is not a problem. Getting the list of all supported commands is. One cannot just run `-h`.

jakub-uC, 2020-09-30:
@eugmes  I could implement this within the `help` command. It could list all available root commands.
Will it be ok?

eugmes, 2020-09-30:
@jakub-uC: yes, that would be great.

PR #28835 (2020-10-02): Shell enhance help command
To get a list of all supported shell commands user must hit the TAB button. It may be awkward on some terminal emulators.
This change enhances the help command in a way it will in addition print all supported commands.

Fixes: #28785
commit f43985e3 (2020-10-02): shell: enhance help command

To get a list of all supported shell commands user must hit the TAB
button. It may be awkward on some terminal emulators. This change
enhances the help command in a way it will in addition print all
supported commands.

Fixes: #28785
closed 2020-10-02 by nashif