zephyrproject-rtos/zephyr #28787: lwm2m-client sample can't be build with openthread and DTLS

retfie, 2020-09-29:
Attempt to build lwm2m-client example for nrf52840dk_nrf52840 target with enabled OT and DTLS produce warning and abort build.

**To Reproduce**
`west build -s samples/net/lwm2m_client -b nrf52840dk_nrf52840 -- -DCONF_FILE="prj.conf overlay-ot.conf overlay-dtls.conf"`

Following error is print:

warning: MBEDTLS_DTLS (defined at modules/Kconfig.tls-generic:32) has direct dependencies (MBEDTLS_TLS_VERSION_1_1 || MBEDTLS_TLS_VERSION_1_2) && MBEDTLS_BUILTIN && MBEDTLS_CFG_FILE = "config-tls-generic.h" && MBEDTLS with value n, but is currently being y-selected by the following symbols:
 - NET_SOCKETS_ENABLE_DTLS (defined at subsys/net/lib/sockets/Kconfig:57), with value y, direct dependencies NET_SOCKETS_SOCKOPT_TLS && NET_SOCKETS && NETWORKING (value: y), and select condition NET_NATIVE && NET_SOCKETS_SOCKOPT_TLS && NET_SOCKETS && NETWORKING (value: y)

error: Aborting due to Kconfig warnings

CMake Error at /home/ice/work/embedded/wsn/zephyrproject/zephyr/cmake/kconfig.cmake:239 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /home/ice/work/embedded/wsn/zephyrproject/zephyr/cmake/app/boilerplate.cmake:591 (include)
  /home/ice/work/embedded/wsn/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /home/ice/work/embedded/wsn/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:35 (include_boilerplate)
  CMakeLists.txt:4 (find_package)

**Expected behavior**
Build finish successful.

**Impact**
Can't get DTLS working on lwm2m example.

**Logs and console output**

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: gcc-arm-none-eabi-7-2018-q2-update
 - Zephyr version: 2.4.99
 - Commit:  aea9486e87840bd3b2515fb0c1409de485fef0f5

**Additional context**

rlubos, 2020-09-30:
OpenThread overrides the default mbedTLS configuration (i. e. disables TLS since it's not used by OpenThread in basic configuration). Try adding `CONFIG_MBEDTLS_TLS_VERSION_1_2=y` to your `overlay-ot.conf` or `overlay-dtls.conf`.

kpetrov-mms, 2020-09-30:
@rlubos , that fixes the problem. Now I got it to compile successful, I can confirm that device now can connect to Leshan LWM2M server, with DTLS and bootstraping. We should update the docs about lwm2m_client sample in this section.
Thanks !

rlubos, 2020-09-30:
Glad to hear it helped. We should add "official" OT support to the sample in the first place (a verified overlay config for OT). Feel free to contribute if you have time 😄

closed 2020-10-02 by retfie