zephyrproject-rtos/zephyr #28803: Use generic config option for early platfrom init (AKA "warm boot")

abrodkin, 2020-09-30:
On some platforms missing simple true reset capability we may mimic real reset (see https://github.com/zephyrproject-rtos/zephyr/issues/28493) with software emulation of that (AKA "warm boot") when we return some internal platform's states into its defaults (see https://github.com/zephyrproject-rtos/zephyr/pull/28494). That among other things allows us to efficiently use the same always powered-on board for continuous tests execution etc.

Currently that kind of functionality is treated as a very board/platform/CPU/architecture specific. And while implementation is indeed typically is very specific for each and every platform the concept itself is quite generic as well as has the same features:
* It's still quite far from the real "cold boot" as we only selectively recover some internal states: some we just don't touch yet, some we cannot even reach to.
* In real products "warm boot" is barely needed and might even introduce more problems re-initializing pristine out-of-reset HW

So that might be a good idea to move that kind of functionality under the same umbrella config option so that we:
1. Document it properly once and for all
2. Have an easy and clean way to disable all that nonsense for production use

ioannisg, 2020-09-30:
Thanks for opening the ticket @abrodkin.
A had a quick look in the Cortex-M architecture and it seems that the following HW need re-initialization, when Zephyr boots as a chain-loadable image (not via reset) and we need to guarantee true reset conditions:

- Main stack pointer (MSP) and CONTROL register
- MPU system
- FPU control/status registers
- Cache clean up and invalidate registers
- Interrupt-related registers (PRIMASK, FAULTMASK, BASEPRI)
- Stack Pointer limit registers, when present
- Debug unit
- External interrupt controller (NVIC)

Re-initializing the above makes Zephyr _deterministic_ and  more _robust_ when it is chain-loadable, and also removes the need for state cleanup at the parent image. But it adds to code footprint, so it should not be done if reset-conditions are guaranteed.

Agree, the idea is to put all this under the same umbrella Kconfig switch so we have a better overview of this.

nvlsianpu, 2020-10-07:
https://github.com/JuulLabs-OSS/mcuboot/issues/833 is related to this ticket.

ioannisg, 2020-10-21:
Relates also to #7404

abrodkin, 2021-05-18:
@ioannisg is there anything else to be done here or we may close it now?

ioannisg, 2021-05-18:
I think we can close this. We already have introduced config INIT_ARCH_HW_AT_BOOT, and this is already used in ARC and ARM architecture, so I'd say we're done.

closed 2021-05-18 by abrodkin