zephyrproject-rtos/zephyr #28819: memory order and consistency promises for Zephyr atomic API?

pabigot, 2020-09-30:
The [Linux atomic operations](https://www.kernel.org/doc/html/latest/core-api/atomic_ops.html) document memory order behavior: in particular, while all are SMP safe, only the functions that return a value also have the necessary memory barriers to guarantee consistent ordering between threads (read-only and write-only operations do not have barriers).

The C11 `<stdatomic.h>` API also defines [memory order](https://en.cppreference.com/w/c/atomic/memory_order) and consistency behavior.

The [Zephyr API](https://docs.zephyrproject.org/latest/reference/kernel/other/atomic.html?highlight=atomic#atomic-services) specifies:

> Using the kernel’s atomic APIs to manipulate an atomic variable guarantees that the desired operation occurs correctly, even if higher priority contexts also manipulate the same variable.

I own my pedantry, but "correctly" doesn't precisely identify the memory order semantics.

What is the promised memory order behavior for Zephyr's atomic functions, which provide RMW, RO, and WO behavior?

pabigot, 2020-10-04:
A specific example of what needs to be clarified:

If in one thread I use `atomic_set(&pending, 1)` and then another thread possibly on another core uses `atomic_cas(&pending, 1, 0)` is it guaranteed the second thread will see the value as 1 (assuming no other access intervenes)?

If the answer is "no", which I think would be the case for Linux, would changing the first to `atomic_cas(&pending, 0, 1)` change the answer to "yes"?

andrewboie, 2020-10-04:
Nearly all systems, including all of our SMP-enabled arches, use the GCC builtins with `__ATOMIC_SEQ_CST` which is documented as "Enforces total ordering with all other __ATOMIC_SEQ_CST operations".

https://gcc.gnu.org/onlinedocs/gcc/_005f_005fatomic-Builtins.html

I haven't looked at how Linux implements atomic ops, I try to avoid looking at Linux source as much as I can for obvious reasons.

Certain Xtensa CPUs that aren't cache coherent between cores might not, @andyross knows more about this than I do, although this has implications for much more than just atomic operations.

pabigot, 2021-02-03:
The fix has been waiting for review and merge for 2 x 60 days already.

pabigot, 2021-04-05:
Make that six months.

closed 2021-07-08 by github-actions[bot]