zephyrproject-rtos/zephyr #28833: STM32: SPI DMA Driver - HW CS handling not compatible with spi_nor (Winbond W25Q)

LuxLuthor, 2020-10-01:
**Describe the bug**
When spi_nor driver is sending command over SPI it uses two rx buffers and two tx buffers. In current implementation in function transceive_dma (spi_ll_stm32.c) SPI peripheral is turned off after each buffer finishes resulting in CS pin being pulled up between bytes send. Flash that I use (Winbond W25Q) after falling edge on CS expect command to be send therefore communication does not work. 

**To Reproduce**
Steps to reproduce the behavior:
1. Always reproducible when using spi_nor driver with spi_dma turned on and hardware slave select
CONFIG_SPI_NOR=y
CONFIG_SPI_STM32_USE_HW_SS=y
CONFIG_SPI_STM32_DMA=n

**Expected behavior**
CS pulled not being pulled up when switching DMA for next buffers. This is how it behaves when software CS is used. 

**Impact**
Prevents SPI flash chips form being used with DMA

lowlander, 2020-10-05:
@LuxLuthor what SOC are you using ?

LuxLuthor, 2020-10-05:
> @LuxLuthor what SOC are you using ?

I use ST32WB55RG

lowlander, 2020-10-05:
> I use STM32WB55RG

Hmm one of the only few STM SOC's I don't have :-/

But I assume the problem happens on all STM32 SOC's, I am not sure if it can be fixed, will have to dive deeper into the DMA/SPI documentation to see how it can be done.

>Impact Prevents SPI flash chips form being used with DMA

You mean not being able to use DMA + HW CS, right ? 

For your use case would using "software" CS be a workaround, or is there a special reason you want/need "hardware" CS support?

LuxLuthor, 2020-10-05:
I would guess that this is visible on every STM32. Using software CS would be a workaround but I looked into driver code and moved turning SPI peripheral ON and OFF outside while loop that is handling iteration over buffers. It seems to work well now. I will make pull request later today.

lowlander, 2020-10-05:
> moved turning SPI peripheral ON and OFF outside while loop that is handling iteration over buffers.

We will really have to double check if that doesn't have any side effects, the DMA timing is a bit tricky.  But we can discus that in your PR.

LuxLuthor, 2020-10-05:
I removed only disabling SPI peripheral. TXDMAEN RXDMAEN stayed in place. Tested on WB55RG with Winbond W25Q and works fine. Enabling/disabling TXDMAEN RXDMAEN while SPI enabled does not seem to be against manual.

PR #28920 (2020-10-15): SPI: Do not disable SPI peripheral while switching DMA buffers
Removed SPI peripheral disabling when switching DMA to another buffer. When using hardware chip select this would cause to stop driving CS pin when swhitching buffers. This is different (and wrong) than when used software CS.

Fixes #28833
commit f091aab0 (2020-10-15): SPI: Do not disable SPI peripheral while switching DMA buffers

Removed SPI peripheral disabling when switching DMA to another buffer.
When using hardware chip select this would cause to stop driving
CS pin when swhitching buffers. This is different (and wrong) than
when used software CS.

Fixes #28833
closed 2020-10-15 by ioannisg