zephyrproject-rtos/zephyr #28844: Double quote prepended when exporting CMAKE compile option using zephyr_get_compile_options_for_lang()

kurddt, 2020-10-01:
**Describe the bug**
A extra and unwanted '"' is prepended when using the output of zephyr_get_compile_options_for_lang() to set an other target compile option. This lead to a compilation failure (gcc does not really like this extra double quote)

example:
```
/usr/bin/arm-none-eabi-gcc -DLZG_SAVED_WINDOWS=1 -O3 -funroll-loops -W -Wall -Wno-extra " -O2 -imacros /home/glager/node/mcu/TN2/bluid/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -imacros /home/glager/node/mcu/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -ffunction-sections -Wformat=0 -Wno-format-security -fno-reorder-functions -fno-defer-pop -fstack-usage -ffunction-sections -fdata-sections -specs=nano.specs -std=c99" -MD -MT liblzg/src/lib/CMakeFiles/lzg.dir/checksum.c.obj -MF liblzg/src/lib/CMakeFiles/lzg.dir/checksum.c.obj.d -o liblzg/src/lib/CMakeFiles/lzg.dir/checksum.c.obj -c /home/glager/node/mcu/tremornet.tools/liblzg/src/lib/checksum.c
```
instead of
```
/usr/bin/arm-none-eabi-gcc -DLZG_SAVED_WINDOWS=1 -O3 -funroll-loops -W -Wall -Wno-extra -O2 -imacros /home/glager/node/mcu/TN2/bluid/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -imacros /home/glager/node/mcu/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -ffunction-sections -Wformat=0 -Wno-format-security -fno-reorder-functions -fno-defer-pop -fstack-usage -ffunction-sections -fdata-sections -specs=nano.specs -std=c99" -MD -MT liblzg/src/lib/CMakeFiles/lzg.dir/checksum.c.obj -MF liblzg/src/lib/CMakeFiles/lzg.dir/checksum.c.obj.d -o liblzg/src/lib/CMakeFiles/lzg.dir/checksum.c.obj -c /home/glager/node/mcu/tremornet.tools/liblzg/src/lib/checksum.c
```

**To Reproduce**
* Have a zephyr-independant target (here lzg) and do:
```
zephyr_get_system_include_directories_for_lang(C ext_system_includes)
zephyr_get_compile_definitions_for_lang(C ext_definitions)

zephyr_get_compile_options_for_lang(C ext_options)

target_compile_options(lzg PRIVATE ${ext_options})
target_include_directories(lzg PRIVATE ${ext_system_includes})
```

Only `zephyr_get_compile_options_for_lang` seems to add this extra double quote, `zephyr_get_system_include_directories_for_lang` does not

**Expected behavior**
The extra double quote should not be added

**Impact**
This is kind of a show stopper since I can't compile my application anymore 

**Environment (please complete the following information):**
 - OS: Linux
 - GNU Arm Embedded toolchain 
 - Zephyr-v2.4.0

tejlmand, 2020-10-01:
The `zephyr_get_compile_options_for_lang(C ext_options)` was originally intended to be used integration to external build system/libraries, such as here:
https://github.com/zephyrproject-rtos/zephyr/blob/49fdfd4f39d5dc575319cf52a054791b22aac7be/samples/application_development/external_lib/CMakeLists.txt#L55

but it does have a side-effect, and that is that it is process order dependent (you may not get flags added after this code was called).
I have done some cleanup and improvements, but need to keep the code working for main use case.
A safer approach in your case is to do:
```
target_compile_options(lzg PRIVATE $<TARGET_PROPERTY:zephyr_interface,INTERFACE_COMPILE_OPTIONS>)
target_include_directories(lzg PRIVATE $<TARGET_PROPERTY:zephyr_interface,INTERFACE_SYSTEM_INCLUDE_DIRECTORIES>)
```

also look at openthread for inspiration:
https://github.com/zephyrproject-rtos/openthread/blob/zephyr/zephyr/CMakeLists.txt#L214-L229

btw.  your code snippet doesn't use: `ext_definitions`

kurddt, 2020-10-02:
Ok, thanks for the tip. This is working perfectly

closed 2020-10-02 by kurddt