zephyrproject-rtos/zephyr #28847: code_relocation sample does not work on windows

KevinThompson-SR, 2020-10-01:
code_relocation sample doesn't work on windows (paths in Windows includes ":", which causes problems).

**To Reproduce**
Steps to reproduce the behavior:
1. /zephyr/samples/application_development/code_relocation example
2. nrf5340pdk_nrf5340
3. using .proj and CMakeList.txt
4. Build Error:  Occurs because the result of line.split(':') contains three elements (["'SRAM2", 'C', '/code_relocation/src/test_file1.c']) and it tries to assign it to two variables (see here).

**Expected behavior**
Relocate desired regions to SRAM.

**Impact**
CRITICAL: Required for final test of our product... We move all regions to SRAM during manufacturing test 

other (build):
Booting: *** Booting Zephyr OS build v2.3.0-rc1-ncs2 ***
using: nrf5340pdk_nrf5340_cpunet

**Logs and console output**
build will fail: Segger Embedded Studio for ARM

**Environment (please complete the following information):**
 - OS: Windows 10
 - nRF Connect SDK v1.3.1

**Additional context**
Modifying gen_relocate_app.py caused other errors... I think it would be wise for the domain expert to take this one on and provide a fix.... would be greatly appreciated!  Best - Kevin

jturnsek, 2020-12-07:
The same with mimxrt1064_evk board.

tejlmand, 2020-12-08:
@KevinThompson-SR @jturnsek  Fixed here: #30528

PR #30528 (2021-01-20): cmake: scripts: supporting gen_relocate_app feature in windows
Fixes: #28847

This commit fixes two places that was causing gen_relocate_app to fail
in windows.

gen_relocate_app.py now splits only on first `:` in
`<MEM_REGION>:<file>`.

Windows contains `:` in path, thus only first `:` is valid for splitting
mem region and file path.

Second part of the issue is fixed in CMake where `'` was used for
quoting of command arguments.
This causes a file not found on Windows because the final `'` would be
treated as part of the filename. Similar the first `'` would be treated
as path of the mem regi [... cut]
commit 1bd0b299 (2021-01-20): cmake: scripts: supporting gen_relocate_app feature in windows

Fixes: #28847

This commit fixes two places that was causing gen_relocate_app to fail
in windows.

gen_relocate_app.py now splits only on first `:` in
`<MEM_REGION>:<file>`.

Windows contains `:` in path, thus only first `:` is valid for splitting
mem region and file path.

Second part of the issue is fixed in CMake where `'` was used for
quoting of command arguments.
This causes a file not found on Windows because the final `'` would be
treated as part of the filename. Similar the first `'` would be treated
as path of the mem reg [... cut]
closed 2021-01-20 by carlescufi