zephyrproject-rtos/zephyr #28854: `CONFIG_STACK_POINTER_RANDOM` may be undefined

jjmilburn, 2020-10-02:
**Describe the bug**

In some builds, it may be possible for `CONFIG_STACK_POINTER_RANDOM` to be undefined. This will trigger a warning (leading to an error if `Werror` is present) in GCC when `Wundef` is set.

**To Reproduce**

Seen using PlatformIO, have no reason to believe that this is related to the PIO code, however.

Steps to reproduce the behavior:
1. Prepare the the 'blinky' sample project for [Nucleo-F103RB](https://docs.zephyrproject.org/1.14.1/samples/basic/blinky/README.html#blinky-sample). 

2. Include the following in prj.conf (may not be necessary, just including for reproducibility):
```
CONFIG_GPIO=y
CONFIG_CONSOLE_SUBSYS=y
CONFIG_CONSOLE_GETLINE=y
```
 
3. Add following build flags:
```
    -Werror
    -Wall
    -Wextra
    -Warray-bounds
    -Wno-sign-conversion  # Zephyr
    -Wlogical-op
    -Wformat-overflow
    -Wformat=2
    # -Wshadow  # Zephyr
    -Wstringop-overflow
    -Wswitch-default
    -Wundef
    -Wno-unknown-pragmas
    -fstack-protector-strong
```
4. Attempt to build image using preferred toolchain (reproduced using PlatformIO build, should be reproducible using ninja/west as well)

**Expected behavior**
No warnings in the Zephyr kernel code due to `Wundef`

Observe warnings similar to the following:
```
/home/j/.platformio/packages/framework-zephyr/kernel/include/kernel_internal.h:122:5: error: "CONFIG_STACK_POINTER_RANDOM" is not defined, evaluates to 0 [-Werror=undef]
```

**Impact**

Need to disable warnings that would normally be enabled as blocking failures in our codebase, or explicitly exclude Zephyr files from analysis.

**Logs and console output**

(Includes other warnings, which might be filed as separate issues if this is an appropriate format)
```
/home/j/.platformio/packages/framework-zephyr/lib/libc/minimal/source/stdlib/exit.c: In function '_exit':
In file included from /home/j/.platformio/packages/framework-zephyr/kernel/include/ksched.h:11,
                 from /home/j/.platformio/packages/framework-zephyr/arch/arm/core/aarch32/cortex_m/thread_abort.c:22:
/home/j/.platformio/packages/framework-zephyr/kernel/include/kernel_internal.h:122:5: error: "CONFIG_STACK_POINTER_RANDOM" is not defined, evaluates to 0 [-Werror=undef]
 #if CONFIG_STACK_POINTER_RANDOM
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/j/.platformio/packages/framework-zephyr/lib/libc/minimal/source/stdlib/exit.c:10:16: error: unused parameter 'status' [-Werror=unused-parameter]
 void _exit(int status)
            ~~~~^~~~~~
cc1: all warnings being treated as errors
Compiling .pio/build/nucleo_f103rb/arch__arm__core__aarch32__cortex_m/zephyr/arch/arch/arm/core/aarch32/cortex_m/irq_init.c.o
*** [.pio/build/nucleo_f103rb/lib__libc__minimal/zephyr/lib/libc/minimal/source/stdlib/exit.c.o] Error 1
cc1: all warnings being treated as errors
*** [.pio/build/nucleo_f103rb/arch__arm__core__aarch32__cortex_m/zephyr/arch/arch/arm/core/aarch32/cortex_m/thread_abort.c.o] Error 1
/home/j/.platformio/packages/framework-zephyr/kernel/idle.c: In function 'set_kernel_idle_time_in_ticks':
/home/j/.platformio/packages/framework-zephyr/kernel/idle.c:66:49: error: unused parameter 'ticks' [-Werror=unused-parameter]
 static void set_kernel_idle_time_in_ticks(s32_t ticks)
                                           ~~~~~~^~~~~
/home/j/.platformio/packages/framework-zephyr/kernel/idle.c: In function 'z_sys_power_save_idle_exit':
/home/j/.platformio/packages/framework-zephyr/kernel/idle.c:116:39: error: unused parameter 'ticks' [-Werror=unused-parameter]
 void z_sys_power_save_idle_exit(s32_t ticks)
                                 ~~~~~~^~~~~
/home/j/.platformio/packages/framework-zephyr/lib/libc/minimal/source/stdlib/atoi.c: In function 'atoi':
/home/j/.platformio/packages/framework-zephyr/lib/libc/minimal/source/stdlib/atoi.c:37:2: error: switch missing default case [-Werror=switch-default]
  switch (*s) {
  ^~~~~~
In file included from /home/j/.platformio/packages/framework-zephyr/kernel/include/ksched.h:11,
                 from /home/j/.platformio/packages/framework-zephyr/kernel/sem.c:27:
/home/j/.platformio/packages/framework-zephyr/kernel/include/kernel_internal.h:122:5: error: "CONFIG_STACK_POINTER_RANDOM" is not defined, evaluates to 0 [-Werror=undef]
 #if CONFIG_STACK_POINTER_RANDOM
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
*** [.pio/build/nucleo_f103rb/kernel/zephyr/kernel/idle.c.o] Error 1
cc1: all warnings being treated as errors
*** [.pio/build/nucleo_f103rb/lib__libc__minimal/zephyr/lib/libc/minimal/source/stdlib/atoi.c.o] Error 1
cc1: all warnings being treated as errors
*** [.pio/build/nucleo_f103rb/kernel/zephyr/kernel/sem.c.o] Error 1
In file included from /home/j/.platformio/packages/framework-zephyr/kernel/include/ksched.h:11,
                 from /home/j/.platformio/packages/framework-zephyr/kernel/mem_slab.c:14:
/home/j/.platformio/packages/framework-zephyr/kernel/include/kernel_internal.h:122:5: error: "CONFIG_STACK_POINTER_RANDOM" is not defined, evaluates to 0 [-Werror=undef]
 #if CONFIG_STACK_POINTER_RANDOM
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
```

**Environment (please complete the following information):**
 - OS: Linux (Ubuntu 18)
 - Toolchain: PlatformIO Core 5.0.1/Home 3.3.0
 - Zephyr Version 2.3.0

nashif, 2020-10-06:
We probably should not be using `#if` here and instead use #if defined()

jjmilburn, 2020-10-07:
I'm happy to push a trivial PR with something like:

```
#if defined(CONFIG_STACK_POINTER_RANDOM)
```

or

```
#if defined(CONFIG_STACK_POINTER_RANDOM) && CONFIG_STACK_POINTER_RANDOM == 1
```

if preferred.

jjmilburn, 2020-10-15:
@nashif PR is here: https://github.com/zephyrproject-rtos/zephyr/pull/29208/

closed 2020-12-29 by github-actions[bot]