zephyrproject-rtos/zephyr #28864: sanitycheck: Make sanitycheck test specifiacation compatible

PerMac, 2020-10-02:
As a tester, I would like to be able to extract the complete and exact test specification (the list of all the test suites with all the test cases) that will be run by sanitycheck for a given setup. This is an important feature in testing standards, which allows verifying the test results against the specification prepared in advance. Currently, there exist some options that allow exporting tests in advance. However, there are still mismatches between the tests in the extracted list and the final report (due to various reasons). One can see such mismatch between the specification and the results on the `tests/subsys/settings/functional/` example:
To get specification, call:
```
./scripts/sanitycheck -T tests/subsys/settings/functional/ -p native_posix --export-tests out.txt
```
To execute the tests and obtain results, call:
```
./scripts/sanitycheck -T tests/subsys/settings/functional/ -p native_posix 
```
The results are found by default in `./sanity-out/sanitycheck_report.xml`. 

I believe this enhancement can be a good goal for v2.5 release. Being able to compare the results with the specification prepared before the test execution is an important feature of QA. 

I would also consider adding a verification step for new tests being added, during which "the specification compatibility" is checked.

nashif, 2020-10-22:
do you see this as an enhancement to sanitycheck or to the tests? Looks like the example you list above is not compliant and does things differently than all other tests, so it should be fixed or adapted.

PerMac, 2020-10-22:
I think there are a couple of tests with similar issues (being not compliant). Another example is cmsis_dsp tests which also cause a big mismatch. We should be able to track such tests down.  
The best would be to fix them without looking for workarounds on the sanitycheck level. So in that matter, it is an enhancement to the tests. 
But I would also like to see some stricter approach for specification compatibility requirement, so we can claim that zephyr testing (both tests and sanitycheck script) fulfills this standard. Creating a test verifying if the extracted specification match 1:1 test cases in the results could lead us to this goal. We could use it to track down non-compliant tests and work on fixing them and also use it as a required check before a new test is added. 
The proposed goal could be fulfilled when e.g. all the cases in the results of tests available for some selected platforms are checked against their specification and 1:1 match is obtained.

closed 2022-11-19 by nashif