zephyrproject-rtos/zephyr #28865: Doc: Generate documentation using dts bindings

erwango, 2020-10-02:
**Is your enhancement proposal related to a problem? Please describe.**
dts bindings provide useful details on:
- what is implemented
- how it is implemented
- how it could/should be configured
Similarly to Kconfig and API it would be quite interesting to generate documentation out of dts/bindings.

**Describe the solution you'd like**
A doc section (ending up being available on https://docs.zephyrproject.org/) that would provide all details on available bindings.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Example where bindings could be useful:
Looking to drivers/sensors  we can't find driver for lis2dh12. Fact is that lis2dh12 is supported by lis2dh driver, which is a piece of information hard to find out.
Looking at the bindings, we can see straight away that `st,lis2dh12-i2c.yaml` is available and hence supported.

erwango, 2020-10-02:
@carlescufi, @MaureenHelm, @galak, @nashif

erwango, 2020-10-02:
Logging the point as it has been discussed at several occurrences already.

avisconti, 2020-10-02:
@erwango 
Thx for the initiative. The lisdh12 device case happened recently. It seems to be not supported, but it is indeed (by lis2dh). 
I was thinking more to a document listing all compatibles string, but I guess this bindings idea is not too different.
Indeed we need this information somewhere.

mbolivar-nordic, 2020-10-02:
@erwango I can try to help with this. My current priorities are TBD right now but I'll self-assign so I don't lose track of this. I'd like to work on usability improvements along with new features.

avisconti, 2020-10-15:
@mbolivar 
When do you think you can start working on it.
And do you have any specific idea already in mind?
Thanks!

mbolivar-nordic, 2020-10-15:
> @mbolivar
> When do you think you can start working on it.

I have started work

> And do you have any specific idea already in mind?

Yes, but it's requiring a significant refactor to edtlib. 

The main problem is that there is no concept of a 'binding' as a standalone object. Bindings in edtlib are currently private instance data of the EDT object. But you can only construct an EDT object if you have a DTS file, which isn't the case for this issue, where we're just querying the file system for available bindings independent of any DTS.

I'm getting closer to ripping that all into a separate Binding class that will be a new top level entry point to the library but my attention this week is a bit divide. I hope to have a PR posted this week that does this refactor at least and we can continue the documentation feature enablement on top of that.

PR #29374 (2020-11-02): Generate documentation using dts bindings
This pull request extends the devicetree reference documentation to include information on each available binding.

Fixes: #28865
commit 808dd9e6 (2020-11-02): doc: reference: add devicetree bindings content

Add generated documentation content for known devicetree bindings
using the Binding abstraction which was just added to edtlib. This
works similarly to the way Kconfig content is generated, so extract a
bit of common helper code for doing that out and rename the relevant
files to keep the distinction clear.

Make the documentation build system respect a preset DTS_ROOT. In this
way, out of tree bindings can be added to the generated content by
telling the documentation build system where to find them, identically
to how out of tree bindings can  [... cut]
closed 2020-11-02 by nashif