zephyrproject-rtos/zephyr #28872: Bluetooth Host only does not work with nucleo_f411re

Cimex97, 2020-10-02:
**Describe the bug**
I want to build the samples/bluetooth/peripheral_hr as a host only with an extern esp32 as a controller. 
For this I use the **nucleo_f411re** board. 

```
*** Booting Zephyr OS build zephyr-v2.4.0  ***
ASSERTION FAIL [err == 0] @ WEST_TOPDIR/zephyr/subsys/bluetooth/host/hci_core.c:375
        k_sem_take failed with err -11
[00:00:10.014,000] <err> os: r0/a1:  0x00000003  r1/a2:  0x0000000a  r2/a3:  0xffffffbf
[00:00:10.014,000] <err> os: r3/a4:  0x08001761 r12/ip:  0x00000000 r14/lr:  0x080037d9
[00:00:10.014,000] <err> os:  xpsr:  0x61000000
[00:00:10.014,000] <err> os: Faulting instruction address (r15/pc): 0x080037e4
[00:00:10.014,000] <err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
[00:00:10.014,000] <err> os: Current thread: 0x20000da8 (unknown)
[00:00:10.059,000] <err> os: Halting system

```

**To Reproduce**
I add hw-flow-control to the dts, the PINMUX functions for RTS/CTS pins of the UART_1 and set the configuration BT_UART_ON_DEV_NAME="UART_1".
```c
&usart1 {
	current-speed = <115200>;
	status = "okay";
	hw-flow-control;
};
```
```c
static const struct pin_config pinconf[] = {
#if DT_NODE_HAS_STATUS(DT_NODELABEL(usart1), okay) && CONFIG_SERIAL
	{STM32_PIN_PB6, STM32F4_PINMUX_FUNC_PB6_USART1_TX},
	{STM32_PIN_PB7, STM32F4_PINMUX_FUNC_PB7_USART1_RX},
	{STM32_PIN_PA11, STM32F4_PINMUX_FUNC_PA11_USART1_CTS},
	{STM32_PIN_PA12, STM32F4_PINMUX_FUNC_PA12_USART1_RTS},
#endif
```

Steps to reproduce the behavior:
1. cd samples/bluetooth/peripheral_hr 
2. west build -b nucle_f411re
3. west flash
4. See error

**Expected behavior**
I expected that I can use Bluetooth specific APIs in my application if I connect an external Bluetooth controller, like the esp32.

**Impact**
I can't implement my application, because I want to use Zephyr, instead of the [esp-idf](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/) for the hardware independent.

**Logs and console output**
Here is logic analyzer trace: 
[image]
I see that there is a communication, but I don't know why the RTS/CTS pins don't change the state.

**Environment (please complete the following information):**
 - OS: linux
 - zephyr 2.4.0
 -  arm-none-eabi-gcc v10.2
 - board: nucleo_f411re
 - bluetooth controller: esp32 with a hci_uart interface 

**Additional context**
On the ESP32 run this [sample](https://github.com/espressif/esp-idf/tree/8bc19ba893e5544d571a753d82b44a84799b94b1/examples/bluetooth/hci/controller_hci_uart) which make a uart_hci interface available. This works fine with an raspberry pi 2b+.

erwango, 2020-10-05:
@Cimex97 Hard to reproduce w/o the matching HW. Can you enable CONFIG_BT_DEBUG_HCI_DRIVER ?

Cimex97, 2020-10-05:
Here is the output with `CONFIG_BT_DEBUG_HCI_DRIVER=y`
```
*** Booting Zephyr OS build zephyr-v2.4.0  ***
[00:00:00.004,000] <dbg> bt_driver.h4_open: 
[00:00:00.004,000] <dbg> bt_driver.rx_thread: started
[00:00:00.004,000] <dbg> bt_driver.rx_thread: rx.buf 0x00000000
[00:00:00.004,000] <dbg> bt_driver.h4_send: buf 0x20003370 type 0 len 3
ASSERTION FAIL [err == 0] @ WEST_TOPDIR/zephyr/subsys/bluetooth/host/hci_core.c:375
        k_sem_take failed with err -11
[00:00:10.014,000] <err> os: r0/a1:  0x00000003  r1/a2:  0x0000000a  r2/a3:  0xffffffbf
[00:00:10.014,000] <err> os: r3/a4:  0x08001761 r12/ip:  0x00000000 r14/lr:  0x080039d1
[00:00:10.014,000] <err> os:  xpsr:  0x61000000
[00:00:10.014,000] <err> os: Faulting instruction address (r15/pc): 0x080039dc
[00:00:10.014,000] <err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
[00:00:10.014,000] <err> os: Current thread: 0x20000da8 (unknown)
[00:00:10.059,000] <err> os: Halting system
```

erwango, 2020-11-04:
@Cimex97 genuine question: ESP32 is supported on Zephyr as a WIFI controller, but is that supposed to work in BLE mode ?

erwango, 2020-11-04:
> @Cimex97 genuine question: ESP32 is supported on Zephyr as a WIFI controller, but is that supposed to work in BLE mode ?

@jhedberg maybe you can answer this question?

Cimex97, 2020-11-05:
Zephyr don't support the ESP32 with BLE mode. This wrote @carlescufi in this [issue](https://github.com/zephyrproject-rtos/zephyr/issues/27055#issuecomment-662537537).

But in my case, zephyr doesn't run on the ESP32.

erwango, 2020-11-05:
@Cimex97 ok, so we can close this issue then, right ?

Cimex97, 2020-11-05:
Maybe not. I want to use a nucleof411re board as a Bluetooth host and the ESP32 as a Bluetooth controller. 

The ESP32 is connected over UART to the nucleof411re and I want to use the hci_uart interface to have Bluetooth support on this board.

erwango, 2020-11-05:
> I want to use a nucleof411re board as a Bluetooth host and the ESP32 as a Bluetooth controller.

So maybe it could work, but in that case, since there is no official support for that, this is not a bug but a feature request and might require some development.

carlescufi, 2020-11-13:
> Zephyr don't support the ESP32 with BLE mode. This wrote @carlescufi in this [issue](https://github.com/zephyrproject-rtos/zephyr/issues/27055#issuecomment-662537537).
> 
> But in my case, zephyr doesn't run on the ESP32.

Is there a build that acts as a controller over UART for the ESP32? Or can you make one? If you can come up with one, then using Zephyr as the Host is trivial and requires only configuration.

Cimex97, 2020-11-16:
I have running this [sample](https://github.com/espressif/esp-idf/tree/8bc19ba893e5544d571a753d82b44a84799b94b1/examples/bluetooth/hci/controller_hci_uart) on the ESP32 and can it use as a Bluetoothcontroler. This is come from the [ESP-IDF](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/index.html).

 This works fine when I build a Zephyr sample for the **native_posix_64** board. But this doesn't work if I build it for the **nucleo_f411re**.

erwango, 2020-11-20:
@Cimex97, would you mind providing mode information on the sample configurations you're using (native_posix_64 vs nucleo_f411re).

Right now I don't have ESP32, so I can't test and see what does (/not) happen.
Though, I'm comparing uart drivers, and IIC, native_posix_64 doesn't handle "hw-flow-control".
Are you sure hw-flow-control is expected on ESP32 side ? Have you tried running the sample w/o this option ?

I'm sorry for asking these questions, but w/o available HW to reproduce, my ability to support is limited.

Cimex97, 2020-11-23:
I'm sure that the ESP32 need hw-flow-control. This is required for the ESP32 as a UART HCI controller.

I tested it with the `samples/bluetooth/peripheral_hr` (same as above). I only deactivate `BT_HCI_ACL_FLOW_CONTROL=n`. Maybe this isn't supported by the ESP32 HCI stack. If it is enabled I get this warning: `[00:00:00.300,000] <wrn> bt_hci_core: opcode 0x0c33 status 0x11` and this error `Bluetooth init failed (err -5)`.

This is the configuration of the sample:
```
CONFIG_BT=y
CONFIG_BT_DEBUG_LOG=y
CONFIG_BT_SMP=y
CONFIG_BT_PERIPHERAL=y
CONFIG_BT_DIS=y
CONFIG_BT_DIS_PNP=n
CONFIG_BT_BAS=y
CONFIG_BT_HRS=y
CONFIG_BT_DEVICE_NAME="Zephyr Heartrate Sensor"
CONFIG_BT_DEVICE_APPEARANCE=833

BT_HCI_ACL_FLOW_CONTROL=n
```
I build and flash it onto the nucleo_f411re and I got the error message above. If I build it for native_posix_64, then I can start the application with `sudo build/zephyr/zephyr.exe --bt-dev=hci0`. Here the ESP32 is with a USB <-> UART converter attached to my Linux system.

erwango, 2020-11-23:
@Cimex97, please note that BT_HCI_ACL_FLOW_CONTROL, is HCI level flow control, which is different from the RTS/CTS uart level flow control.

What I'm proposing is to have a try w/o the UART level flow control (`&usart1 {hw-flow-control;};`), as these pins don"t seem to be handled by native_posix driver. This being said, this is a naive try, as native_posix is not a basic uart driver.
Anyway, I should hopefully be able to get an ESP32 this week.

Cimex97, 2020-11-23:
Ok I tried to disable the hw-flow-controll. 
This works for the natvie_posix_64. I attached the device with `btattach -B /dev/ttyUSB1 -S 115200 -P h4 -N`, where `-N` means no hw-flow-contoll.

On the nucle_f411re I get now a lot of this error messages:
` [00:00:00.009,000] <err> bt_driver: TX interrupt but no pending buffer!`

**Note:** I have to connect the CTS pin from the ESP32 to the RTS-pin. I don't know why.

erwango, 2020-11-23:
>  I have to connect the CTS pin from the ESP32 to the RTS-pin

That's how CTS/RTS works.

> On the nucle_f411re I get now a lot of this error messages:
`[00:00:00.009,000] <err> bt_driver: TX interrupt but no pending buffer!`

Can you try to increase BT_RX_STACK_SIZE, say CONFIG_BT_RX_STACK_SIZE=2048 ?

Cimex97, 2020-11-23:
Than I get this message again:
```
*** Booting Zephyr OS build zephyr-v2.4.0-586-g4d0750920b52  ***
ASSERTION FAIL [err == 0] @ WEST_TOPDIR/zephyr/subsys/bluetooth/host/hci_core.c:376
        k_sem_take failed with err -11
[00:00:10.015,000] <err> os: r0/a1:  0x00000003  r1/a2:  0x0000000a  r2/a3:  0xffffffbf
[00:00:10.015,000] <err> os: r3/a4:  0x080016c5 r12/ip:  0x00000000 r14/lr:  0x08003839
[00:00:10.015,000] <err> os:  xpsr:  0x61000000
[00:00:10.015,000] <err> os: Faulting instruction address (r15/pc): 0x08003844
[00:00:10.015,000] <err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
[00:00:10.015,000] <err> os: Current thread: 0x20000d80 (unknown)
[00:00:10.060,000] <err> os: Halting system
````

erwango, 2020-11-23:
ok, txs.
I'll really need to get a way to test this myself. I'll get an ESP32 and get back to you.

Cimex97, 2020-12-04:
Hey @erwango, 
do you have some new information about this issue?
If I could help you to just ask me :).

erwango, 2020-12-04:
Hi @Cimex97, sorry finally I haven't got a chance to find a ESP32 (was a ESP8266, so no bt...). I need to order one.
So I'm actually sorry, but w/o way to reproduce for now, it will be difficult to help more, at  least for my part.

Cimex97, 2020-12-04:
Okay, thank you for your answer.

erwango, 2021-01-08:
@Cimex97 not sure it could help, but since it could have an impact on the API behavior, maybe you could have a try to https://github.com/zephyrproject-rtos/zephyr/pull/31192

carlescufi, 2021-01-12:
@Cimex97 would you be able to try this again with the latest master now that #31192 is merged?

Cimex97, 2021-01-14:
Hey @carlescufi, 
I will do it in a few weeks.

Cimex97, 2021-02-01:
Hey, I update now to master. I got the same error.

carlescufi, 2021-02-01:
@Cimex97 that failing `k_sem_take` you are getting is due to the controller not replying to a command. As to why it is not responding, the most likely reason is a UART driver problem. Could you post a full logic analyzer trace of the whole exchange, and, if possible, a transcription of the trace as bytes going to the controller and coming from the controller?

Cimex97, 2021-02-03:
I traced it with the logic analyzer and realized that the ESP32 does not answer. After I enable the hw-flow-control and set the CTS/RTS pins in the device tree it just works. Many thanks for your support. 
```
pinctrl-0 = <&usart1_tx_pb6 &usart1_rx_pb7 &usart1_rts_pa12 &usart1_cts_pa11>;
```

**Note**: I have to disable *BT_HCI_ACL_FLOW_CONTROL*. Maybe the ESP32 does not support that.

erwango, 2021-02-03:
@Cimex97 Thanks for this news! Let me know if I can close the issue.

closed 2021-02-03 by erwango

dogtopus, 2021-08-08:
@Cimex97 I think flow control is supported by ESP32. However there are 2 issues that prevents it from working:

1. The ACL MTU without header must be >= 1017 (i.e. ACL packet length must >= 1021). This is a known quirk as noted by IDF devs [here](https://github.com/espressif/esp-idf/issues/942) and easily worked around by just increase the L2CAP MTU in Zephyr.
2. The SCO MTU and count probably needs to be non-zero. I wasn't able to verify this because for some very weird reason my ESP32 fails to even respond to Reset command (returns error code 0x80) when I set dummy values for `hbs->sco_mtu` and `hbs->sco_pkts` in `set_flow_control` in Zephyr (remind you that I run Zephyr on a separate processor and all requests are flowing through UART so it shouldn't affect HCI in any way :confused: ).

With these 2 solved we probably could get the flow control working but I still have no idea what's wrong with case 2.

mirko, 2022-12-11:
> Ah, I see. Thank you. ESP's implementation has a bug which bothers me, so I'm gonna search further. I thought Zephyr can also work as a HCI controller firmware :)

Can it (by now)? I'm sorta lost about whether I can use an ESP32 on linux using basic HCI or not by today.

danergo, 2022-12-11:
You don't need zephyr, you can just use esp's hci_uart_sample (or similar,
its in their repo). I could fine-tune it and proved to work for quite a
long time already.

danergo, 2022-12-11:
> Interesting, thanks! Didn't realise they implemented the HCI stack
> complete enough to be usable with blueZ.
> Does it work with a normal cheap node-mcu pcd and their CH9102X UART<->USB
> bridges? Asking, because I read that RTS/CTS is a necessity for attaching
> it as UART driven BLE-HCI peripheral.

I don't know, I'm using it with full featured UART to SPI chips, as HCI
needs proper CTS, RTS, and of course RX and TX and GND.