zephyrproject-rtos/zephyr #28893: Double-dot in path's may cause problems with gcc under Windows

AdrianEggenberger, 2020-10-03:
**Describe the bug**
In our project that is using Zephyr 2.4.0 in combination with GCC for ARM under Windows we struggled over an issue that was totally confusing. In our CMakeFiles.txt we set some additional paths for SO, BOARD and DTS. Based on our directory structure we had paths with double-dots in it (e.g. "c:/myroot/main/../device/board"). The project could be configured with cmake, but when the Ninja was triggered to build the project a re-run of cmake is triggered. This ends in an endless loop with re-run of the cmake.

After a lot of reading and studying the CMake/Ninja functionality I found out that the re-run is caused by an entry in the ninja.build file inputs. Somehow a "c$:" entry was added to the input files that always triggered the re-run.

My first thought was that it's a bug in cmake, but after updating to the latest version and additionally debugging cmake based on the git repository I cam to the conclusion that the unexpected dependency 'c$:' comes from the zephyr environment.

In the next step I found that the includes in my <BOARD>.dts file are somehow the source for the issue. Looking at the <BOARD>.dts.pre.d file in the build folder showed the source of the problem:
`empty_file.o: \
 C:/Daten/git/topic-mcuboot/zephyr/misc/empty_file.c \
 C:/Daten/git/topic-mcuboot/device/boards/arm/myboard/myboard.dts \
 c:\daten\git\topic-mcuboot\device\dts\arm\silabs\efr32fg14p231f128gm32.dtsi \
 C:/Daten/git/topic-mcuboot/zephyr/dts/common/mem.h`

The paths of files that are in a folder that is added by the double-dots have a backslash notation and the others have a forward slash.  This file is afterwards split into the different dependency files by a split based on backslash's in dts.cmake. The path with the backslashes is into multiple elements and also a "c:". 

If the paths for the board, soc and dts are given without double-dots all files are given with forward slashes.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a project with additional board folder that contains double-dots and add a custom board.
2. Start the build by "west build"
3. See the invalid backslashes in the *.dts.pre.d file in the build/zephyr folder

**Expected behavior**
Double-dots should work as other path variants.

**Impact**
The issue could be fixed by not using double-dots, but it took me a full day to get the reason for the strange behavior.

**Environment (please complete the following information):**
 - OS: Windows)
 - Toolchain (e.g Zephyr SDK, ...): Zephyr 2.4.0, CMake 3.18.3, GNU for ARM 2020-q2

tejlmand, 2020-10-05:
@AdrianEggenberger This is related to this:
https://github.com/zephyrproject-rtos/zephyr/issues/26112

> In our CMakeFiles.txt we set some additional paths for SO, BOARD and DTS. Based on our directory structure we had paths with double-dots in it (e.g. "c:/myroot/main/../device/board"). 

Please ensure your paths are CMake style by using:
```
file(TO_CMAKE_PATH "${my_path}" cmake_path_file)
```
see more here:
https://github.com/zephyrproject-rtos/zephyr/pull/27839

tejlmand, 2020-10-27:
Fixed here: https://github.com/zephyrproject-rtos/zephyr/pull/29578

commit ad787661 (2020-10-27): cmake: dts: getting realpath for system includes for dts.

Fixes: #28893

Now using get_filename_component(REALPATH) to ensure full path with
symlinks and `..` resolved.

Having `..` in path causes the gcc generated dependency file to contain
`\` in path on windows. This confuses ninja.
The result is that ninja consider the file missing, and thus invokes a
CMake run to create the file, resulting in endless loop.

By ensuring that full path (containing `/` which is already ensured)
when calling gcc, then all paths will be posix style, making ninja
happy.
closed 2020-10-27 by carlescufi