zephyrproject-rtos/zephyr #28897: SPI does not work for STM32 min dev board

ppeetteerrs, 2020-10-04:
**Describe the bug**
Using Zephyr's SPI driver, I am unable to write / read bytes to the SPI slave. The default port mappings are used. With the same configuration, the Arduino framework works perfectly fine.

**To Reproduce**

This is my `main.cpp` and `prj.conf`
```cpp
// main.cpp
#include <string.h>

#include <zephyr.h>
#include <device.h>

#include <drivers/gpio.h>
#include <drivers/spi.h>

static struct device *gpio_dev;

static struct device *spi;

static struct spi_config spi_cfg;

static struct spi_buf_set spi_buf_set_tx;
static struct spi_buf_set spi_buf_set_rx;

static struct spi_buf spi_buf_tx;
static struct spi_buf spi_buf_rx;

static uint8_t tx_buf[255];
static uint8_t rx_buf[255];

int main(void)
{
	k_msleep(3000);
	gpio_dev = device_get_binding("GPIOA");
	printk("stupid u\n");

	spi = device_get_binding("SPI_1");
	if (!spi)
	{
		printk("Could not find SPI driver\n");
		return -1;
	}
	printk("SPI: 0x%X\n", (unsigned int)spi);

	spi_cfg.operation = SPI_WORD_SET(8);
	spi_cfg.frequency = 125000;

	spi_buf_tx.buf = tx_buf;
	spi_buf_tx.len = 5;

	spi_buf_rx.buf = rx_buf;
	spi_buf_rx.len = 5;

	spi_buf_set_tx.buffers = &spi_buf_tx;
	spi_buf_set_tx.count = 1;

	spi_buf_set_rx.buffers = &spi_buf_rx;
	spi_buf_set_rx.count = 1;

	while (1)
	{
		memset(&tx_buf[0], 0, 255);
		memset(&rx_buf[0], 0xFE, 255);

		int r1 = spi_transceive(spi, &spi_cfg, &spi_buf_set_tx, &spi_buf_set_rx);

		printk("%d \n", r1);
		for (int i = 0; i < 10; i++)
		{
			printk("0x%x ", *(rx_buf + i));
		}

		k_msleep(1000);
	}
}
```

```sh
# prj.conf
CONFIG_STDOUT_CONSOLE=y
CONFIG_PRINTK=y

CONFIG_I2C=y
CONFIG_SPI=y
CONFIG_GPIO=y
CONFIG_ASSERT=y
CONFIG_SPI_STM32_USE_HW_SS=y
CONFIG_PINMUX=y
```

**Expected behavior**
The code is expected to run smoothly. 

**Logs and console output** 
The Serial print shows an error code of -22. I have tried the sample code on the Zephyr github and it shows the same error.

**Environment (please complete the following information):**
Dell XPS 17, Windows 10. I am using Zephyr on PlatformIO.

lowlander, 2020-10-05:
@ppeetteerrs can you try `CONFIG_SPI_STM32_USE_HW_SS=n` ?

lowlander, 2020-10-05:
@ppeetteerrs and what SOC are you using ?

ppeetteerrs, 2020-10-05:
Hi, I tried with `CONFIG_SPI_STM32_USE_HW_SS=n` and it still shows error code -22. I have attached my `platformio.ini` below:
```
[env:zephyr]
platform = ststm32
board = bluepill_f103c8
framework = zephyr
upload_protocol = serial
monitor_speed = 115200
```

ppeetteerrs, 2020-10-05:
Adding `spi_cfg.slave = 1` also doesn't help.

ppeetteerrs, 2020-10-05:
Okay, apparently defining the arrays using `static uint8_t* tx_buf;` solved the issue. Managed to solve this thanks to one line of comment saying we should pass arrays as pointers inside the documentation for syscalls.

ppeetteerrs, 2020-10-05:
Sorry, this is not fixed actually.

ppeetteerrs, 2020-10-06:
Okay, I figured out its the frequency issue. Not sure why too low of a frequency causes the error.

closed 2020-10-06 by ppeetteerrs

VividhaDhengre-eaton, 2025-02-25:
@ppeetteerrs At What frequency your code is working, because for us as well the SPI in not working for stm32g0 series, using zephyr